digistump/OakCore #1: Clean up and Integrate OakParticle library directly into core

digistump, 2016-01-15:
## Part 1
Refactor OakParticle.h and particle_core.h to be a properly formatted set of classes, with proper header and code files:

No functional changes should be made to the code, any functional changes should be separate pull requests after the code is cleaned up. 

The end result should be two exposed classes:
Particle (currently in OakParticle.h) - which should expose the methods that it currently exposes. No changes needed to how these are exposed, but they need to be tied into the code in particle_core.h so that is within the class (or a second class that Particle uses) and not publicly exposed. None of the variables or methods from particle_core.h should be publicly exposed except those exposed by the Particle class and the Oak class (detailed next).

Oak - which is a new class that should expose the following methods from particle_core.h:
rebootToUser();
rebootToConfig();

particle_core.h needs to be split into at the least, a header and code file, and/or more divided up further and/or combined with OakParticle.h to achieve the above goals. My lack of C++ expertise really shows up here, help with this would be most appreciated.

## Part 2
The OakParticle library and presumably the ESP8266WiFi library (as the OakParticle library requires it) as well, need to be directly included into the core and included in Arduino.h so that the Particle class methods are available both in the sketch (without any includes directly in sketch) and in the core's main.cpp file (cores/esp8266/core_esp8266_main.cpp)

## How to test
Install the latest stable ESP8266 core 2.0.0 (https://github.com/esp8266/Arduino) via board manager
Download this repo and place in OakParticle folder in libraries.
Complete Part 1, ensuring that this example compiles: [ParticleTest.zip](https://github.com/digistump/OakCore/files/91369/ParticleTest.zip)

Download this zip: [core_esp8266_main.zip](https://github.com/digistump/OakCore/files/91308/core_esp8266_main.zip)
Complete Part 2 - test by doing the following:

Replace the (cores/esp8266/core_esp8266_main.cpp) in the ESP8266 core with the core_esp8266_main.cpp in the zip. This file calls the Particle class and expects it to be included through Arduino.h. Ensure that the example sketch in this zip (calls the Particle class without including it directly, since it should be included in the core) compiles with that new core_esp8266_main.cpp in place.

You will not yet be able to test on a device, but since no functional changes are required for this, compiling without errors is enough and we will take it from there.

## BOUNTY
Reward: Must complete both parts - $300 cash or $450 credit at Digistump.com or 50 Oaks

Deadline: 11:59 PM 2015-01-15 (Friday January 15th) - the first pull request that meets all requirements will win the bounty. Additional improvements after that may or may not be rewarded, depending on the contribution, at our discretion. 

If you are sure you can do this, strong in your C/C++ abilities, and able to do this quickly - please feel free to respond to this issue and say that you are working on it and when you will complete it, so that others don't waste their time on it. If you are just entertaining the idea/unsure if you can do it, please don't "claim" it until you are sure.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. The winner will be the first pull request that submits fully working code meeting the above requirements and following good coding practices, based on the timestamp of the pull request. Bounty will be awarded (or in the case of Oaks, sent) within 48 hours of confirming winner. Cash awards will be made in USD. This is not an offer for hire.  All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

danielmawhirter, 2016-01-15:
Compiling the example (ParticleTest.ino) sketch leaves it complaining about c++11 support on dsakeygen.cpp:25. This is strange, because in %APPDATA%\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\platform.txt, compiler.cpp.flags includes -std=c++11.
"Does this work on your machine?"
I am working on the refactoring, but I make no claim to be able to finish until I am able to compile.

DarkLotus, 2016-01-15:
Im trying to get the Sample sketch compiling before attempting any changes. But im getting errors about Cache_Read_Enable_New() being already defined, yet to hit c++11 issues.

digistump, 2016-01-15:
@danielmawhirter - I get warnings about c++11 macros requirements when I compile, but that is it. In case it matters I'm compiling on windows - but it looks like you are too. I will try a full clean install on a different machine first thing in the morning to see if I can reproduce those errors, but given that I don't know much about the compiler flags I'm quite sure I haven't messed with them.

digistump, 2016-01-15:
@DarkLotus sorry about that - I long forgot that I had weakened that method in libmain in the SDK. That isn't needed for this code to compile or work so I've removed the file that causes that error if it isn't weakened - delete "oakboot-bigflash.c" or update to the latest commit (which does the same) and it should go away. Would love to hear if you then see c++11 errors like @danielmawhirter or not?

Thanks to you both for taking a look!

DarkLotus, 2016-01-15:
No problems, compile's fine once removed no sign of any c++11 errors. Using Arduino 1.6.5 on Mint 17.3.

DeuxVis, 2016-01-15:
I've started to work on this, but the deadline is too close for me to succeed in time, and I need some rest.

By the way for which timezone is this ?

fri-sch, 2016-01-15:
I refactored the classes for part 1 (https://github.com/fri-sch/OakCore).
But it's not clean as I have no more time in the next few hours.

danielmawhirter, 2016-01-15:
@fri-sch You got through a few things that were causing me trouble. If sharing the bounty is possible, I will work from that starting point, as I do have time for the next few hours.

DeuxVis, 2016-01-15:
I'm giving up you guys are faster than me - that was fun to try ;)

@digistump I understand you're under some pressure to release a firmware for the oak, but at some not too far point in the future I'd recommend having the code redone from scratch if you can afford it. Maintainability will be a beast with this inherited "glue" coding you had this started from.

digistump, 2016-01-15:
@DeuxVis, @fri-sch, @danielmawhirter, @DarkLotus  Thank you all for working on this! 

@DeuxVis the timezone was Pacific time I guess, but I would have taken it by that time in any timezone, given the amount of C++ foo required I wasn't sure if it was realistic but figured it was worth a shot

@fri-sch Please email me at support@digistump.com, I'd like to send you a small bounty for not just working on it, but posting your code for others to work from - that is afterall what open source is all about!

@DeuxVis I'll be happy to have someone write this into something more maintainable - this is essentially the Particle code, which is a huge beast of abstraction layers, distilled into something I can understand and maintain (if a bit painfully) - I hope to progress from there while preserving the methods that are "untouched" from Particle so they can easily be kept compatible. 

If anyone sees any major improvements to be made on the code @povrazor submitted please feel free, happy to give away things like Oaks for minor updates.

@povrazor - Thank you! I will test it all in the next few hours, but I'm closing this issue because from a first look it seems you've done the impossible, made my heap of code into something slightly more coherent and include-able. Please email me at support@digistump.com and let me know what bounty choice you'd like, and we'll go from there.

Thank you all again!

danielmawhirter, 2016-01-15:
@digistump In the current pull request, the particle-core functions are still globally defined. Since that is the one major thing I managed to fix while I was working on it, I'm integrating it into @provrazor 's code to submit an additional pull request

povrazor, 2016-01-15:
Thanks @digistump :)

@danielmawhirter They shouldn't be. ```particle_core.h``` is only included by the cpp files (Oak.cpp and OakParticle.cpp).

digistump, 2016-01-15:
A few remaining changes left on this are open to anyone, especially @povrazor (the original pull requester), @fri-sch and @danielmawhirter (since understand exactly what is needed/have done it).

See #2

digistump, 2016-01-15:
Another c/c++ bounty posted @ #3

povrazor, 2016-01-15:
Easy enough. Here's a small patch that moves everything from ```particle_core``` in to a namespace.

povrazor, 2016-01-15:
I changed my mind on what the namespace should be. All the internal code tends to use an ```underscore_syntax```, so I switched it from my original ```ParticleCore``` namespace to ```particle_core```.

https://github.com/digistump/OakCore/pull/5

digistump, 2016-01-16:
Thanks again @povrazor - closing this.

closed 2016-01-16 by digistump