digistump/OakCore #7: BOUNTY: Write Shield Tutorials

digistump, 2016-01-17:
This is in progress, check back soon

TheDarkTerminal, 2016-01-17:
Hey Erik,

I am very keen on your products and have already received my oak. I would love to participate with writing up tutorials for shields. I have professional lighting equipment and a Canon DSLR, a lot of knowledge about shields for Arduinos and other products such as yours. I selected a cheaper backer plan as before trying out shields I wanted an actual bade product first and see if I would be interested in buying shields for it later if I liked it. I would be more than happy to help you write a couple tutorials however I don't own the shields just yet. I was wondering if you would like to send me a shield or two and then I would be able to write up tutorials for that? The shield itself would be my bounty and nothing else. Up to you of course, I'd love to help though!

Thanks,

Tom

digistump, 2016-01-18:
@TheDarkTerminal - thanks for the awesome offer - no doubt an arrangement like that would be ideal for  getting some very high quality tutorials when things aren't so urgent. The two main priorities for these though, is getting tutorials up quickly (otherwise I'll do them in the next day or three) and engaging the community as much as possible. I'd love to take you up on your offer in the future, but for these tutorials I want to first open them up to people who have shields and are ready to go, and hopefully therefore open them up to several people who are interested as well. Thanks!

TheDarkTerminal, 2016-01-18:
Ah sure thing Erik! I had thought of urgency as being a drawback to my offer but if you want some high quality tutorials done in the future for possible new shields that you might bring out then feel free to mail one and I'll get it done for you! Working to create communities and working with them for an awesome product is enough of a bounty for me, but writing tutorials would be great as I've a big passion for electronics like these.

Thanks again and super excited for the firmware release, WiFi setup is looking good! (I took a little snoop),

Tom

briankb, 2016-01-18:
Erik, I have the beta kit and will do the proto shield tutorial. I'm assuming it will use a lot of the existing content for the Digispark Proto shield as they will be almost identical except for the PCB.

Is there something specific you want done with the Oak Proto Shield other than the male headers?

briankb, 2016-01-18:
Oak Prototype Shield tutorial completed, photos are attached. Hopefully the image names and links on the wiki page are correct, they should be.
[Oak_Proto_Shield_Pics.zip](https://github.com/digistump/OakCore/files/93779/Oak_Proto_Shield_Pics.zip)

digistump, 2016-01-18:
@briankb - Excellent, looks great - thank you! I've removed the Proto Shield from the list, and you have claimed the bounty. Please email me at support@digistump.com to claim it and let me know if you want Oaks or Credit - if you plan to do more of these then feel free to email me after you've done all you plan to and I can then give you one big credit or shipment of Oaks (or some of both)

Thank you for your always awesome support as well!

@TheDarkTerminal - thank you again for the offer - I fully intend to take you up on it! And thanks for your support!

TheDarkTerminal, 2016-01-24:
@digistump Hey Erik,

Just a reminder in case you still need shield tutorials done. I'm still offering my offer and would love to help make some tutorials. My Oak board is fully functioning now and if you'd like to send me a shield or two, I could get the tutorials done within a couple days of receiving them :)

Hope your family is well, and great job on the firmware!

Tom

exeng, 2016-02-03:
Erik, Built two Oak Level Shifters. Have a tutorial draft ready with pics. Ready to write a level shifter wiki tutorial.
Steve

exeng, 2016-02-03:
Erik, Just completed the initial (for review) Oak Level Shifter Shield tutorial. First time Wiki'r but managed to figure it out. Have to thank briankb for provided the first example. It help with tables and nested link syntax. Let me know if this works for you or if you need anything else. Please review. T and M image files are attached. Got bit by the 10MB limit. I'll have to piecewise send ZIPs of the large images.
[Oak-Level-Shifter-Shield-t-m.zip](https://github.com/digistump/OakCore/files/115215/Oak-Level-Shifter-Shield-t-m.zip)
Steve

exeng, 2016-02-03:
Erik, Here is 1 of 3 large image ZIPs. Sorry for the large number of images but there were 23 photo in the tutorial.
Steve
[Oak-Level-Shifter-Shield-l-1.zip](https://github.com/digistump/OakCore/files/115248/Oak-Level-Shifter-Shield-l-1.zip)

exeng, 2016-02-03:
Erik, 2 of 3...
[Oak-Level-Shifter-Shield-l-2.zip](https://github.com/digistump/OakCore/files/115251/Oak-Level-Shifter-Shield-l-2.zip)

exeng, 2016-02-03:
Erik, Last one 3 of 3. Let me know if you need anything else. Good night.
Steve
[Oak-Level-Shifter-Shield-l-3.zip](https://github.com/digistump/OakCore/files/115255/Oak-Level-Shifter-Shield-l-3.zip)

exeng, 2016-02-04:
Erik, Noticed that you edited my Oak Level Shifter Shield tutorial submission. You changed the parts list to include call out 1x40 pin header, 47 pins worth. Will there be 2 1x40 headers in the kit? Perhaps broken as you suggest? That seems to be what happen to my kits. Wondered why there was a different mix of headers from one kit to the next.
Steve

exeng, 2016-02-04:
Erik, Took another look at the Oak Level Shifter Shield tutorial. Noticed that the step numbers jump from 6 to 8. This was my bad. In my original offline draft I had a step 6 and 7 that was combined for dokuwiki version and I forgot to renumber accordingly. Basically 6 and 7 were separate steps (before combined) for the 2 and 3 screw terminals. If you want me to fix the numbering, let me know. Else, I'll assume you'll fix this before you officially release the turtorial for use.
Steve

3dpinta, 2016-02-05:
I'll be getting the TFT shield tutorial up later tonight. Thanks!

3dpinta, 2016-02-06:
Sorry it was over 6 hours but I have it done. Here's the tutorial.
[Oak TFT Shield](http://digistump.com/wiki/oak/tutorials/tft)
And the pictures.
[Oak TFT Shield.zip](https://github.com/digistump/OakCore/files/120124/Oak.TFT.Shield.zip)

exeng, 2016-02-07:
Erik, Corrected step number sequence in Level Shifter Shield tutorial.
Steve

inventhouse, 2016-02-11:
Erik, I built the Weather Shield last night and took pictures, though I haven't tested it (beta firmware, no 3v3 programmer); I'll try to write it up this evening.

inventhouse, 2016-02-11:
Erik, I can't find the "reset password" on the wiki; I was 'bjh' there before I got crazy-busy...

inventhouse, 2016-02-11:
Got images cropped and annotated and some words written last night, hopefully I can bang out the rest tonight.

inventhouse, 2016-02-12:
Erik, For "medium" images should I aim for dimensions that approximate 854x480 pixels (e.g. a square image should be scaled to sqrt(854x480) = 640x640)?

inventhouse, 2016-02-12:
Anyway, I just have to scale the images and transfer the writeup to the wiki once I get back in; Erik, for the password reset, feel free to contact me at the email associated with my 'bjh' wiki account.

inventhouse, 2016-02-15:
Here's my writeup in wiki format with images: 
[WeatherBuild.zip](https://github.com/digistump/OakCore/files/131593/WeatherBuild.zip)

3dpinta, 2016-02-16:
Erik, I took the writeup from bjh/inventhouse and put it up on the wiki so you won't have to worry about doing at least that minimal part.
I'll have to get myself a weather shield now too!

digistump, 2016-02-20:
@inventhouse - please send me an email at support@digistump.com to claim your credit.

Thanks to everyone who submitted a shield so far!

wminarik, 2016-02-22:
I'll be uploading the 3.3V Relay Shield tutorial shortly.

wminarik, 2016-02-22:
The 3.3V Relay Shield tutorial has been added. Tiny and medium images are attached here, originals will be mailed to support@ since the zip is over 10 MB.
[t.zip](https://github.com/digistump/OakCore/files/140102/t.zip)
[m.zip](https://github.com/digistump/OakCore/files/140103/m.zip)

3dpinta, 2016-02-22:
I'll be posting up the GPS and OLED shield tutorials tonight. If I get a chance, I'll be putting the nRF24L01+ shield up too.

3dpinta, 2016-02-23:
Erik,

So, slight change of plans. I finished the GPS, OLED, and BTLE tutorials tonight instead of the nRF24L01+ shield. The pictures are too large to post and to keep things short, you can get them from my [Dropbox](https://www.dropbox.com/s/qm0x8aw9yy14zpx/3xTutorial_Images.zip?dl=0).

For these three shields, I also noticed there was a warning from the Pro tutorials that there were warnings to check with a DMM about shorts to ground. I didn't realize that until after I had soldered all my boards up but I don't know if they apply to these boards. If so, then it's there. If not, it's easy to remove.

I will try to get the nRF24L01+ shield tomorrow and possibly the BT shield if I have time for that too.

3dpinta, 2016-02-25:
I will be finishing up the nRF24L01+ and BT shields tonight.

3dpinta, 2016-02-26:
Here are the pictures for the nRF24L01+ and BT shield tutorials. [Dropbox](https://www.dropbox.com/s/s8at1giahaqn3mt/nRF%20and%20BT.zip?dl=0)

exeng, 2016-03-16:
Updated the level shifter shield tutorial to reflect the pin reversal for GND and P1 on the center servo header. Here is an image for the DokuWiki showing the correct pin positions for GND and P1. Large, Medium and Thumbnail images provided.
[Oak-Level-Shifter-Shield-24-t-m-l.zip](https://github.com/digistump/OakCore/files/175232/Oak-Level-Shifter-Shield-24-t-m-l.zip)

exeng, 2016-03-22:
Did a minor edit to the Temperature Sensor (1-Wire) Shield tutorial to correct the hyperlinks in the Advanced Usage with the Oak section recently added. These were broken.

With regard to the image place holders. What needs to be done? These were uploaded as zip at Bounty #39. They could just a easily be removed without effecting the tutorial.

exeng, 2016-03-22:
@digistump Erik, 
[Oak-Temp-Sensing-1-Wire_Img.zip](https://github.com/digistump/OakCore/files/185156/Oak-Temp-Sensing-1-Wire_Img.zip)
Here are the corrected images for the Temperature Sensor (1-Wire) Shield tutorial Advanced example/tutorial. If you want me to place them directly in the Wiki let me know. Placeholders are in the Wiki.
exeng

digistump, 2016-03-23:
Thanks'd be great @exeng if you can add them directly to the wiki - thanks!

exeng, 2016-03-23:
@digistump Erik, the images have been placed in Wiki.
BTW have you seen the kit Temperature Sensor tutortial/lesson? Finished it last night. Hopefully it meets that lesson template guidelines.
Each should be complete now. Let me know if you anything else.

intheorystudios, 2016-03-24:
Eric, As promised, I just finished the Making your own Oak Smoke Detector+ tutorial.  And here are the images.  Hope I did everything right. Let me know if I need to change anything.  Thanks!
[Oak Smoke Detector Plus.zip](https://github.com/digistump/OakCore/files/187120/Oak.Smoke.Detector.Plus.zip)

zeiv, 2016-03-24:
Hi everyone, I'm working on the LiPo Shield tutorial and will be posting it soon.

zeiv, 2016-03-24:
@digistump Hey Erik, the LiPo Shield tutorial is complete.  I was hoping you could fact-check me with the 6-pin header the board includes, though.  The Oak only needs the 3 VCC/GND/VIN pins as mentioned in the tutorial, right?  I couldn't find any schematics or anything mentioning that the additional pins would provide anything like charge information to the Oak.

exeng, 2016-03-26:
@digistump , et al, I have created a fritzing part for the Oak Level Shifter Shield and recently updated it to reflect the pin swap of GND and P1 on the center servo header. A zip file containing the part is attached.
[OakLevelShifterShieldPart.zip](https://github.com/digistump/OakCore/files/190324/OakLevelShifterShieldPart.zip)

digistump, 2016-03-26:
@zeiv - the 6 pin is just for stability when place on top and/or for when used with stackable headers you can use a Digispark shield on top of it. You are correct about only those three pins in use.

A few other things - that I didn't expect you to guess - but that would be great if you could work them in to your tutorial:

1) The empty two pin holes that aren't part of the 6 or 3 pin header - these are for an optional switch - put a switch or jumper between them and when it is closed power to the Oak will be off.

2) The JST outline on the bottom (shares one common hole with the top one) - put the JST connector in these holes to use it on the bottom, then mount the 3 pin header on top, then you can put the lipo board under the Oak, solder it into the 3 pin, and still use double 9 pin Oak shields on top. 

The tutorial looks great, if you don't mind adding notes for those I'd appreciate it. Please email me to claim your bounty.

Thanks
Erik

Anyone here who hasn't gotten their bounty yet please email me! - lots and lots of bounties so one or two might have got lost in the shuffle.

Also closing this since all kits have tutorials now - see #39 for some mini tutorials that still need doing. Feel free to re-open if you think of a tutorial to do.

closed 2016-03-26 by digistump