digistump/OakCore #11: Build esptool2 for Linux32, Linux64, OSX

digistump, 2016-01-19:
Builds with gcc, no external dependencies - if anyone has a chance to build on these platforms and upload it would be much appreciated.

Clone or download https://github.com/raburton/esptool2
"make"

Upload resulting binary here.

TrueJournals, 2016-01-19:
As requested, for Linux64 :smile: [esptool2.zip](https://github.com/digistump/OakCore/files/95252/esptool2.zip)

digistump, 2016-01-19:
thanks again @TrueJournals

DarkLotus, 2016-01-19:
[esptool2.zip](https://github.com/digistump/OakCore/files/95257/esptool2.zip)
Compiled on osx10.10

DarkLotus, 2016-01-19:
[esptool2-linux32.zip](https://github.com/digistump/OakCore/files/95523/esptool2-linux32.zip)
linux 32 version.

digistump, 2016-01-22:
Thank you both - this was a huge help in getting the first release out tonight!

closed 2016-01-22 by digistump