digistump/OakCore #9: Test Servo Library

digistump, 2016-01-19:


ripred, 2016-01-31:
Erik, once I renamed the older Arduino Servo folder it found and used the esp8266 version.  It initially failed while compiling the esp8266 version of Servo.cpp with errors about min(...) and max(...) being undefined.  I could not get the existing code to work so I added two local functions in my Servo.cpp called getmin(int x, int y) and getmax(int x, int y) and it all compiles and is working for me.  So the only issue seems to be the definition of min and max.   One odd thing to note: After this I went back into the original Arduino/Libraries folder and restored the name of the Servo folder.  I exiting and restarted the Arduino IDE but for some reason it always find the correct version now.  No idea why it didn't revert back to the older warnings/errors but oh well, it works!

ripred, 2016-01-31:
Aha!  I had also removed the (really old) residue Servo folder that used to reside under /Users/USERNAME/Documents/Arduino/Libraries/Servo which I believe was residue from back in the Arduino 1.0.5 days.  So the correct resolution might not be an issue for anyone except really older Arduino users.  

Cheers,
Trent

nokemono42, 2016-02-08:
I'm using OS X and I had a similar problem to @ripred. I ended up adding this to my Servo.h

`#define min(a,b) (((a)<(b))?(a):(b))
#define max(a,b) (((a)>(b))?(a):(b))`

And that seemed to get it working!

ripred, 2016-02-09:
@nokemono42:  Yes that will fix it as well.  I know that on last check Erik had those exact macros commented out from inside the core.  Not sure why other than I suspect they got in the way of something.  Not to mention that macro-based min/max definitions have some possibly bad results such as passing argument expressions that modify existing state such as calling max(x += 1, 0) &c but I'm sure you're aware of these traps.

closed 2016-02-12 by digistump