digistump/OakCore #14: Exec permissions missing from esptool2 on osx

TomKeddie, 2016-01-22:
When compiling on osx with 1.6.6 and oak 0.9.0 I see esptool has the wrong permissions to execute.

`"/Users/tom/Library/Arduino15/packages/digistump/tools/esptool2/0.9.0/esptool2" -quiet -bin -boot2 -4096 -iromchksum "/var/folders/2b/ky7slbm110s6pfjkkfgm6nsh0000gn/T/build98cd95de6f0c5f5844eaaa866b983d69.tmp/blink.ino.elf" "/var/folders/2b/ky7slbm110s6pfjkkfgm6nsh0000gn/T/build98cd95de6f0c5f5844eaaa866b983d69.tmp/blink.ino.bin" .text .data .rodata
fork/exec /Users/tom/Library/Arduino15/packages/digistump/tools/esptool2/0.9.0/esptool2: permission denied
Error compiling.`

`tomk-macbook-pro:vhs-nomos tom$ ls -al /Users/tom/Library/Arduino15/packages/digistump/tools/esptool2/0.9.0/esptool2
-rw-r--r--+ 1 tom  staff  18112 18 Jan 18:39 /Users/tom/Library/Arduino15/packages/digistump/tools/esptool2/0.9.0/esptool2`

Hack fix is to chmod it.

TomKeddie, 2016-01-22:
Same issue for oak binary.

`tomk-macbook-pro:vhs-nomos tom$ ls -al /Users/tom/Library/Arduino15/packages/digistump/tools/oakcli/0.9.0/oak
-rw-r--r--+ 1 tom  staff  19526676 21 Jan 23:19 /Users/tom/Library/Arduino15/packages/digistump/tools/oakcli/0.9.0/oak`

digistump, 2016-01-28:
Fixed in latest release

closed 2016-01-28 by digistump