digistump/OakCore #18: Make wiki about bootloader modes, recovery modes, config mode

digistump, 2016-01-28:
Conditions for entering wifi config:
WiFi configuration is missing or corrupted.
Still cannot connect to WiFi (not Particle, but WiFi) after 5 consecutive tries - aka no internet available.
OTA Update Failed - either the update itself failed, or the newly booted ROM failed.
There is no User ROM or User ROM is invalid/corrupt.
NOT YET IMPLEMENTED: Pin 10 held LOW during boot.

pfeerick, 2016-02-25:
This may not be the best place to discuss this, but I have a question about the behaviour associated with "Still cannot connect to WiFi (not Particle, but WiFi) after 5 consecutive tries - aka no internet available."

Can you make it so that it still retries to connect to it's last remembered AP, and it it is re-established, it jumps back into the user sketch. 

Usage example - we had a 3ish hour power outage due to maintenance to the power lines. Both Oaks went into wifi config mode, and didn't auto-connect back to the configured wifi, as they were running on battery USB and survived the outage. This *could* have still been an issue if they were on wall-warts, as I don't know if the wifi would boot up into time to service the Oaks before they auto-entered wifi config mode. So I had to log into the wifi setup, and reconnect to the AP, and it's back to normal now.

emardee, 2016-02-25:
That was my immediate question too when reading how it now works... sounds like this needs to be a configurable option so we can tailor the behaviour to match our needs.

Also on same train of thought what are the security implications if the Wifi router drops, and therefore our Oaks start broadcasting on their own open wifi to get a new wifi config?

I could imagine scenarios where people would want their Oak to just keep hunting for just the existing secure pre-configured wifi only, and then ONLY re-enter the wifi config routine at a pin held LOW event or tge other valid scenarios given above etc.... but just not during a router / wifi / internet outage.

digistump, 2016-03-23:
This option is added in 1.0.1 coming out now

closed 2016-03-23 by digistump

emardee, 2016-03-23:
Thanks Erik. Look forward to playing with that!