digistump/OakCore #19: Board not recognized by Arduino

Lazarus9000, 2016-01-28:
Finally got one of my units to finish wifi config, updated the Arduino files but if I try to upload is fails with the error below:

Arduino: 1.6.7 (Windows 7), Board: "Oak by Digistump, 80 MHz, Particle OTA, OAK (4M/1M SPIFFS), Single - 1MB (Fullsize)"

Board oak (platform oak, package digistump) is unknown

Error compiling.

digistump, 2016-01-28:
From what I'm reading - This is a known issue with Arduino IDE 1.6.6 and higher, I can confirm it does not always occur nor does it ever occur on 1.6.5. Fastest solution is to delete the oak stuff at:  C:\Users\YOUR_USERNAME\AppData\Roaming\Arduino15\packages\digistump\hardware\oak

closed 2016-01-28 by digistump

eumb, 2016-05-03:
I had a previous oak version. I deleted that 0.9.3 folder and all was working again.