digistump/OakCore #20: Board not registering on Particle cloud

nkildal, 2016-01-28:
Moved from #16 

I did another factory restore of one Oak, then went to http://build.particle.io and removed both oaks.
Followed the procedure at http://rawgit.com/digistump/OakSoftAP/master/config.html, registering a new device. Firmware seemed to download (rapid blinks for minute or so), but the Oak never show up in the Particle cloud and the SoftAP never verified it.
Boot messages on serial (74880 baud) says:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,8
```
I can see that my Oak connects to my wifi and grabs an IP address...

nkildal, 2016-01-28:
Ok - got my Oak into AP mode, by denying its mac address in my Wifi router and rebooting it :-)

Connecting to https://192.168.0.1/particle now gives me:
```
Not connected
```

http://192.168.0.1/info gives me:
```
{"id":"d957040000adbcc2ef4ccaff","claimed":0,"claim_code":"","server_address_type":1,"server_address_domain":"device.spark.io","firmware_version":0,"version_string":"1.0.0","meta_id":0,"meta_data":"","first_update_domain":"oakota.digistump.com","first_update_url":"/firmware/firmware_v1.bin","first_update_fingerprint":"98 66 d5 5c 3d 4a 49 24 e3 1b 72 8b 8f 2e 65 2e 32 2a 7b 95"}
```

digistump, 2016-01-28:
So it is updated - strange 

With the MAC not blocked can you power cycle and then try claiming it via the build.particle.io add device box?

nkildal, 2016-01-28:
I get a "could not claim device":

[image]

nkildal, 2016-01-28:
@digistump: 
Just tried claiming the other Oak - unfortunately same error message...

These 2 Oaks has previously both been registered in the particle cloud - but has both been reset to factory settings, removed from the particle cloud (using remove device in the web interface at build.particle.io)...

digistump, 2016-01-28:
updated bin files and procedures here: https://github.com/digistump/OakRestore/

feel free to test, as it won't hurt anything, but I haven't tested fully yet as I'm on the road at the moment - I'll post back once I get to test tonight.

thanks!

nkildal, 2016-01-28:
Just tested your new restore procedure - it worked!
Great job Erik - all my Oaks are now registered with the particle cloud.
Now - on to testing some Arduino code :smile: 
Thanks a lot!

closed 2016-01-28 by nkildal

DarkLotus, 2016-01-29:
Just wanted to report all working well on my end as well.
Config Wifi, Oak updated, connected to cloud. Pushed a new rom ( took 5 mins 50 seconds  I almost reset it :p ), but all smooth.