digistump/OakCore #21: Arduino Libraries do not download to Arduino IDE

PhillyNJ, 2016-01-29:
Ok, I am following the directions here- > https://github.com/digistump/OakCore/releases/tag/0.9.2 which tells me to also use http://digistump.com/wiki/oak/tutorials/arduino to set up the Arduino Libraries and Boards. I follow the directions. 1st issue: The step says "Select the “Digistump Oak and Acorn” package and click the “Install” button." but its actually called "Oak by Digistump By Digistump" version 0.9.2. Next, no libraries are downloaded but I see the board. I should see an "Oak" folder in my Libraries folder but I do not. Under Examples there is no Oak option. At this point, in particle.io I can connected and see the Oak but no sketch (blink) will compile. I am not sure if this process is ready. I am using a Mac. Please advice.

DarkLotus, 2016-01-29:
Board definitions are not downloaded to your local libraries folder, they go into the system wide libraries folder.

No examples currently bundled but just use the example sketch listed near the bottom on http://digistump.com/wiki/oak/tutorials/arduino

PhillyNJ, 2016-01-29:
Thanks but I get the following error:

"Board oak (platform oak, package digistump) is unknown

Error compiling."

DarkLotus, 2016-01-29:
Which version of OSX are you on if you dont mind me asking?

PhillyNJ, 2016-01-29:
OSX EL Capitan.

DarkLotus, 2016-01-29:
You might need to wipe out your Arduino Libraries folder and begin again.
https://esp8266hints.wordpress.com/2015/12/03/arduino-ide-revision-1-6-6/
Implies this issue is caused if you have ran 1.6.5 than upgraded to 1.6.6 or 1.6.7 which seem to have some incomaptibilites with older settings.

PhillyNJ, 2016-01-29:
There is no way I am wiping out my library. I will throw these oaks in the trash first. As a note, I follow the directions on my windows 10 and I am able to upload a sketch via the IDE... The oak lives another day...

DarkLotus, 2016-01-29:
Glad you got it working on windows.
The issue your seeing is IDE related not just limited to the Oak, You can read yet more about the issue here https://github.com/esp8266/Arduino/issues/973

Seems to be confined to pre 1.6.5 installs upgraded to 1.6.6+

digistump, 2016-01-29:
Just to be completely clear and expand on what @DarkLotus said -

"Board oak (platform oak, package digistump) is unknown" 

That error is caused by a known issue with the Arduino IDE, the fact that the issue is showing up when you install the Oak has nothing to do with the Oak and we can't do anything about it except offer advice as @DarkLotus did. It could happen with any board package you install whether it be from Sparkfun, Adafruit, Digistump, or anyone.  Personally I prefer to use Arduino 1.6.5 as it doesn't seem to suffer from this issue.

closed 2016-01-29 by digistump