digistump/OakCore #22: OneWire Library - #error "Please define I/O register types here"

ajpowell, 2016-01-30:
When compiling code (modified form of supplied example code DS18x20_Temperature - changed to use Particle.delay() rather than delay()), I get the following error:

    Arduino: 1.6.7 (Mac OS X), Board: "Oak by Digistump, 80 MHz, Particle OTA, OAK (4M/1M SPIFFS), Single - 1MB (Fullsize)"
    
    In file included from /Users/apowell/Documents/Arduino/oak_1wire_ds18b20/oak_1wire_ds18b20.ino:1:0:
    /Users/apowell/Library/Arduino15/packages/digistump/hardware/oak/0.9.2/libraries/OneWire/OneWire.h:119:2: error: #error "Please define I/O register types here"
     #error "Please define I/O register types here"
      ^
    exit status 1
    Error compiling.
    
      This report would have more information with
      "Show verbose output during compilation"
      enabled in File > Preferences.

I have looked in the OneWire.h file and see the definitions just prior to line 119, but I do not know what the board is defined as...at a guess, I expect it will match, or be similar to, 'ARDUINO_ARCH_ESP8266'

DarkLotus, 2016-01-31:
I cant find any OneWire devices to test with( i know i have a bunch of DS180's somewhere.... ) but i suspect the standard ESP8266 Arch should work.

If your comfortable trying, edit /packages/digistump/hardware/oak/0.9.2/variants/oak/pins_arduino.h
Insert the following anywhere after line 29, i just stuck it right at the end before the final #endif 
    #define ARDUINO_ARCH_ESP8266 1
This will let you compile, just not sure if it's a full fix though.

ajpowell, 2016-01-31:
Hey @DarkLotus, I tried this - the code compiles and I have deployed OTA - something isn't right though - the serial output is:

    ROM = 28 D9 14 B1 3 0 0 D5
      Chip = DS18B20
      Data = 1 FF 7 4B 46 7F FF 1 10 2F  CRC=2F
      Temperature = 127.94 Celsius, 262.29 Fahrenheit
    No more addresses.

Certainly not 127°C here!  But at least I am getting response from the device!

I will but out all day today - will take a deeper look tonight if not back too late.

DarkLotus, 2016-01-31:
Hey cool it sort of worked, I still cant find a DS18B20, so hopefully someone else can add some more insight.

ajpowell, 2016-01-31:
OK - looked at this again and found that wire connecting to the DS18B20 pin 3 (Vdd) wasn't making a good connection :disappointed:  - replaced the wire and everything is working  :grinning: 

```
ROM = 28 D9 14 B1 3 0 0 D5
  Chip = DS18B20
  Data = 1 6F 1 4B 46 7F FF 1 10 67  CRC=67
  Temperature = 22.94 Celsius, 73.29 Fahrenheit
No more addresses.
```

23°C is more like it for house with central heating on at the end of damp and windy day in England!

digistump, 2016-02-08:
Closing - defining this (in next release) in pins_arduino.h so that it is present for all libraries - thank you both!

closed 2016-02-08 by digistump

Amanrazdan, 2018-11-28:
it was really simple just include the library of the sensor before onewire its fixed ...really stupid

aghosh0605, 2021-09-19:
# Solution
1. If you installed One Wire library from zip then it would be in `C:/Users/<username>/Documents/Arduino/libraries`
2. This path doesn't work for this OneWire library.
3. Delete the *OneWire Folder*
4. Restart Arduino IDE
5.  Go to `tools>manage libraries`
6. Type OneWire in the search box
7. Install that one that has the name ***Paul Stoffregen***
8. Now restart the IDE and try compiling. It will work :)