digistump/OakCore #23: Baud rate of initial serial output from Oak

ajpowell, 2016-01-31:
Would it be possible to change the baud rate of the initial output  on the serial port from the Oak to a 'standard' baud rate?  According to others, they see data at 74880 baud...my serial terminal (Mac OSX CoolTerm) can only monitor at certain rates - 9600, 14400, 28800, 38400, 57600, 115200.

So the initial data, appears to me as something like (seen at 57600 baud):
```
B·
rÄ¤ï¤Bß¨!.Ä
ÇâBóæB)
B(¨¡;«*%ù.¨àòbé¢æ
¦DÃg#y¶ÄÄÄ9ë© .(&bè¢æ
ÆD£ÿÆECbrÞ{}áàÃ.
ÂÂ
```
Others testers have reported output like (seen at 74880 baud):
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,8
```
Could one of the more conventional rates be used? 57600, perhaps?

DarkLotus, 2016-02-01:
As far as i understand it no, The 74880 baud is not changeable as its part of  the esp's actual ROM, not the user portion we modify by flashing the device.

Someone else with more knowledge may be able to answer better but in the mean time you can add a new text file in the same folder as coolterm called 'baudrates.ini' and add 74880 on a single line as explained @ http://forums.the-meiers.org/viewtopic.php?f=4&t=479

ajpowell, 2016-02-01:
Ah!  Thanks @DarkLotus.

OK, if that is essentially fixed/out of our control, then that's that.

As a workaround, changing the available baud rates would work - will try
the baudrates.ini trick when I get home.

ajpowell, 2016-02-02:
BTW - The work around is working nicely - :smile:

digistump, 2016-02-08:
Closing - thanks for answering this @DarkLotus  - your answer was exactly correct.

closed 2016-02-08 by digistump