digistump/OakCore #25: Particle.variable(); with double doesn't compile

digistump, 2016-02-08:
http://digistump.com/board/index.php/topic,1997.0.html

digistump, 2016-02-08:
@DarkLotus @danielmawhirter @povrazor - Would any of you folks with much better C++ foo then me be willing to take a look at this - looking at Particle's original code for this (https://github.com/spark/firmware/blob/35a78512e983e667e3b9a1696396f42b82c66465/wiring/inc/spark_wiring_cloud.h) it seems like it should work but I'm guessing there is a scope/template issue going on here where the linker can't connect it all together...

danielmawhirter, 2016-02-09:
It looks like a version of https://github.com/spark/firmware/blob/35a78512e983e667e3b9a1696396f42b82c66465/wiring/inc/spark_wiring_cloud.h#L134-L137 needs to be added for double? I don't see one in that file.

povrazor, 2016-02-10:
Yeah, there's an Int and String version, but no double.

digistump, 2016-02-11:
Thank you both - that was my thought too - but that doesn't seem to do it (or I'm doing it wrong) in addition the spark code linked works for doubles on particle/spark devices and doesn't have it - my for variables code is essentially a verbatim copy of theirs with some scope changes, so that's where I get lost - I'm going to add a small bounty to this and see if that makes it worth anyone's time to dig in a bit and figure it out (as I'm about to do an update announcing a bunch of bounties)

Thanks again!

digistump, 2016-02-12:
Closing awarded to DarkLotus

closed 2016-02-12 by digistump