digistump/OakCore #28: BOUNTY: Fix hidden network bug

digistump, 2016-02-10:
**Skills Required:** Intermediate at C/C++ and ESP8266 knowledge
**Difficulty:** Unknown

##Challenges/Thoughts:

When in config mode on Windows 8.1 (and likely others) the Oak appears as a hidden network when first powered on (seems to appear as ACORN-XXXXXX after a soft reboot) - you can still connect to it, and so fixing it has not been a high priority previously. My best guess is this involves the order of initializing the Oak as an AP, the scanning code, or the WiFi/AP library itself.

##To test:

- Install Python 2.7 and ensure it is in your path
- Install the latest Oak core via board manager, download this repository and copy over the files found at APP_DATA/packages/digistump/hardware/oak/XXX where APP_DATA is where your system keeps the Arduino data and XXX is the current Oak core version.
- In cores/oak/OakParticle/particle_globals.h uncomment #define OAK_SYSTEM_ROM_4F616B 82 - this tells it to compile the OakSystem sketch as a system image not a user sketch
- Grab the latest OakSystem sketch from https://github.com/digistump/OakSystem
- Make changes to the core and/or sketch to achieve the above task.
- Connect your Oak using a 3.3v USB to Uart adapter, connect Pin2 to GND before powering on to enter Serial bootloader mode (after an upload the code with automatically run and you'll have to power cycle it to go back to bootloader mode. Disconnect Pin2 to GND to have it run the code on next power up)
- ONCE: Grab OakRestore (https://github.com/digistump/OakRestore) and run ```python esptool.py --baud 115200 --port YOUR_COM_PORT write_flash -fs 32m 0x1000 blank.bin 0x2000 oaksetup_restore.bin 0x0081000 oakupdate_restore.bin 0x101000 blank.bin 0x102000 blank.bin 0x202000 blank.bin 
``` replacing YOUR_COM_PORT with your com port, to restore the Oak to factory setup so that the Serial upload will work and the Oak will boot to it.
- Use "Upload Using" "Serial" in the Arduino tools menu. 
- Upload from the IDE and test

## Bounty

$50 cash or $100 credit or 10 Oaks

If you are sure you can do this, and able to do this quickly - please feel free to respond to this issue and say that you are working on it and when you will complete it, so that others don't waste their time on it. If you are just entertaining the idea/unsure if you can do it, please don't "claim" it until you are sure.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. The winner will be the first pull request that submits fully working code meeting the above requirements and following good coding practices, based on the timestamp of the pull request. Bounty will be awarded (or in the case of Oaks, sent) within 48 hours of confirming winner. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

kh90909, 2016-02-14:
I've gone through all the above tesing steps to make sure my Oak is completely up to date, but I haven't been able to replicate this issue on my 64-bit Win 8.1 machine, which has a Broadcom 802.11ac wifi chipset. Same goes for the other devices I tested: a Win 7 laptop and my Android phone.

Might this issue be specific to certain wifi chipsets, or 32-bit Windows?

pfeerick, 2016-02-15:
I would suggest the same - I had no end of trouble trying to get my laptop with intel wireless to connect to the SoftAP for setup, and eventually gave up and connected with the android tablet with no issues whatsover - other than it being a bit slow to get the next configuration page.

thunderace, 2016-03-04:
I can replicate the issue on Win 10 64-bit, but not every time. I'm on it....

thunderace, 2016-03-10:
I made a pull request for OakSystem with a bug fix in the ssid_ap buffer (too small, you need anaother char for the terminal 0).
Perhaps it will fix this issue. To be tested.

digistump, 2016-03-21:
@thunderace - have you tested this any more to see if it was fixed for you? Like you I can't reproduce every time. If you have seen it as fixed and no one else reports it as persisting after 1.0.0 update then you've won this bounty. Please email me at support at digistump dot com and let me know what form of payment you'd like.

Closing, anyone feel free to re-open if you see this persist after updating to 1.0.0

closed 2016-03-21 by digistump

thunderace, 2016-03-22:
Yes, many tests and i can't reproduce it, i think it's fixed. Thank's for the bounty.
I have a new pull request  for the OakSystem (some potential bugs). I'll pull it this week.
Regards.

digistump, 2016-03-22:
Awesome - thanks for all your contributions!