digistump/OakCore #29: BOUNTY: Get FastOTA working

digistump, 2016-02-10:
**Skills Required:** Intermediate at C/C++, Particle knowledge a plus but not necessary
**Difficulty:** Easy to intermediate

## Challenges/Thoughts:

The Particle cloud supports both a slow OTA update mode and a fast mode. The slow mode is currently working with the Oak and works as follows: the cloud breaks the bin file up into chunks and sends them in order one at a time, when the Oak gets the first one it ACKs to the cloud and the cloud sends the next one. Each chunk is 512 bytes. With all of those ACKs this makes for a very long process (2-3 minutes!).
In FastOTA mode the cloud sends the chunks without waiting for ACKs, and then the Oak requests any missing chunks at the end, this should be much faster, but doesn't currently work. The Oak seems to think the first chunk is the last and errors our (or the cloud errors out) we haven't had time to dig into this, hence the bounty!

References: 
Particle code https://github.com/spark/firmware/tree/develop/communication/src
Functions in the particle_core.cpp file that reference update or chunk in their name.
(hint: search the particle code for functions with the same name and usually you'll find their version, which is often very similar)
I'm also happy to facilitate asking questions of the Particle team if needed to solve this.
Using the particle cli tool ```particle subscribe mine``` can give you some good debug info, I can also have Particle turn on additional trace info.

## To test:
- Register and Oak with Particle cloud
- Install Python 2.7 and ensure it is in your path
- Install the latest Oak core via board manager.
- Compile a simple app with a distinct serial and or blinking pattern to use as your test upload, save this bin file somewhere (since you'll be modifying the core files below)
- Download this repository and copy over the files found at APP_DATA/packages/digistump/hardware/oak/XXX where APP_DATA is where your system keeps the Arduino data and XXX is the current Oak core version.
- In cores/oak/OakParticle/particle_globals.h uncomment ```//#define OAK_SYSTEM_ROM_4F616B 82``` - this tells it to compile the OakSystem sketch as a system image not a user sketch - also change ```#define OAK_SYSTEM_VERSION_INTEGER``` to 5 or higher as Particle won't send with FastOTA if the reported version is under 5
- Uncomment ```//#define DEBUG_SETUP``` in that same file, or add your own debug messages (uncommenting that might provide more serial messages then you want)
- Grab the latest OakSystem sketch from https://github.com/digistump/OakSystem
- Make changes to the core and/or sketch to achieve the above task.
- Connect your Oak using a 3.3v USB to Uart adapter, connect Pin2 to GND before powering on to enter Serial bootloader mode (after an upload the code with automatically run and you'll have to power cycle it to go back to bootloader mode. Disconnect Pin2 to GND to have it run the code on next power up)
- ONCE: Grab OakRestore (https://github.com/digistump/OakRestore) and run ```python esptool.py --baud 115200 --port YOUR_COM_PORT write_flash -fs 32m 0x1000 blank.bin 0x2000 oaksetup_restore.bin 0x0081000 oakupdate_restore.bin 0x101000 blank.bin 0x102000 blank.bin 0x202000 blank.bin 
``` replacing YOUR_COM_PORT with your com port, to restore the Oak to factory setup so that the Serial upload will work and the Oak will boot to it.
- Use "Upload Using" "Serial" in the Arduino tools menu. 
- Upload from the IDE
- Test by sending the bin you compiled in the second step to the device using the OakCLI tool

## Bounty

$200 cash or $400 credit or 40 Oaks

If you are sure you can do this, and able to do this quickly - please feel free to respond to this issue and say that you are working on it and when you will complete it, so that others don't waste their time on it. If you are just entertaining the idea/unsure if you can do it, please don't "claim" it until you are sure.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. The winner will be the first pull request that submits fully working code meeting the above requirements and following good coding practices, based on the timestamp of the pull request. Bounty will be awarded (or in the case of Oaks, sent) within 48 hours of confirming winner. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

DarkLotus, 2016-02-13:
Hey, was planning to give this one a shot, but is it just me or does the OakSystem sketch fail to compile? [build log](https://gist.github.com/DarkLotus/aa76bf13b478df95990b)

digistump, 2016-02-13:
A few commits coming in the next hour (and the next beta release) that will
fix that, sorry looks like I made an interim commit at some point.

DarkLotus, 2016-02-13:
ah sweet, didnt think id broken anything late last night hah

digistump, 2016-02-13:
should be good to go now! Beta release 3 is also live for update

DarkLotus, 2016-02-13:
Tried compiling System again, same errors but then i tested [blink with function](https://gist.github.com/DarkLotus/74a52e4acdb619abf28e)
And get similar errors, ie blinkLED is not declared. Seems like the preprocessor is not creating definitions, the sketch in /tmp has the definitions, but not the declarations added at the top like normal.

Adding the definition manually to the top of the sketch compiles fine as expected.

digistump, 2016-02-13:
very strange - no idea why you're getting that, especially with the simple sketch - I just tried a completely fresh install of Arduino 1.6.7 and pulled 0.9.3 via board manager and that worked (as did my previous install of 1.6.5 I've been developing on)

DarkLotus, 2016-02-13:
Ah maybe im going crazy, wiped 0.9.3 and reinstalled again and its now fine.

DarkLotus, 2016-02-14:
[WiP](https://github.com/DarkLotus/OakCore/tree/fastota)
Some work in progress, its after 2am here so im done for the night but i just had my first successful FastOTA flash and it was FAST!

After initally fixing the First chunk issue, next up was it was SLOW still. sending acks. Fixed that and it was Exception(0) after 6-7 chunks.

The final key appears to have been adding a delay.

Will test some more tomorrow morning.

DarkLotus, 2016-02-15:
Another update.

Delay has been tweaked an OTA update takes about 15-30 seconds.

I've ran into a new issue though
[Log](https://www.dropbox.com/s/e7e1r7qrzz1kljv/semiWorking.PNG?dl=0)

The spark events, ie OTA Update Started, do not fire, and particle reports the update as having failed. But as shown in Serial output, the update worked, and rebooted fine.
If i move the spark event OTA Update Started from line 1702 to the start of the function handle_update_begin it fires, but once fastOTA starts it seems like no events are sent out.

DarkLotus, 2016-02-15:
> (04:49:31:pm) <particle-slack> [Matthew McGowan, Particle] Thereâ€™s a small issue with the Cloud that it doesnâ€™t report the correct status of OTA - itâ€™ll on our radar to be fixed this sprint. Sorry for the touble this has caused you.

Looks like it might be an issue on particles end. Ill clean this up into one commit and PR later.

DarkLotus, 2016-02-15:
New branch [fast-ota](https://github.com/DarkLotus/OakCore/commits/fast-ota)

As above just cleaned up and condensed to one commit. I've done 20+ flashes now with no issues apart from particle cloud reporting it failed.

digistump, 2016-02-18:
Awesome! Thank you! I will double confirm with Particle and then merge!

digistump, 2016-02-19:
Yes my contacts at Particle confirm this as well - they will be patching the issue with the failed notice soon, but we are free to ignore it for now - BOUNTY claimed and this will be integrated into the next release - thanks again @DarkLotus

PR #43 (2016-02-19): Fast OTA fixes
Awaiting confirmation that remaining issues are Particle side.
Fixes #29
closed 2016-02-19 by digistump