digistump/OakCore #30: BOUNTY: Debug and create interface for Cloud Serial

digistump, 2016-02-10:
Skills Required: Intermediate at C/C++ and Beginner or higher at Javascript/CSS
Difficulty: Easy to Moderate

## Challenges/Thoughts:

This pull request added the ability to use Particle cloud events like they are a "Serial" object in Arduino: https://github.com/digistump/OakCore/pull/6

- Written data is sent to the cloud as the event oak/device/stdout
- Read data is sent to the Oak via the event oak/device/stdin
- Important messages about the Oaks state are sent to the cloud as the even oak/device/stderr
- System commands can be sent to the Oak via the event oak/device/reset with data "config mode", "user mode", "update mode", "reboot"

(searching these terms in the particle_core.cpp file will give you some insight - ignore #2790-2811 that was replaced by the subscribe at 3058)

**Rate Limiting: **Currently the code checks in the spark_process function for data to send from the buffer, this is called very frequently. There needs to be some kind of limiting functionality as the Particle cloud has a soft limit of 1 event per second per device. So this should be changed so that the buffer contents are sent no more than once every second.

**Debugging: **The results on the cloud seem like they might not be what is expected, this needs to be tested and the code adjusted if necessary. Using the particle cli tool particle subscribe mine can give you some good debug info - worth confirming with output from ParticleJS as well, since that tool can report with strange formatting.

**HTML/JS App: **Create a HTML/Javascript terminal style app to go with this functionality. This is very straightforward using the SparkJS/ParticleJS library. See the attached terminal_example.html for the specs for this.

## To test:

Install Python 2.7 and ensure it is in your path
Install the latest Oak core via board manager, download this repository and copy over the files found at APP_DATA/packages/digistump/hardware/oak/XXX where APP_DATA is where your system keeps the Arduino data and XXX is the current Oak core version.
Use a sketch the utilizes Particle.print, Particle.read, Particle.write
Make changes to the core to achieve the above task.
Connect your Oak using a 3.3v USB to Uart adapter, connect Pin2 to GND before powering on to enter Serial bootloader mode (after an upload the code with automatically run and you'll have to power cycle it to go back to bootloader mode. Disconnect Pin2 to GND to have it run the code on next power up)
ONCE: Grab OakRestore (https://github.com/digistump/OakRestore) and run python esptool.py --baud 115200 --port YOUR_COM_PORT write_flash -fs 32m 0x1000 blank.bin 0x2000 oaksetup_restore.bin 0x0081000 oakupdate_restore.bin 0x101000 blank.bin 0x102000 blank.bin 0x202000 blank.bin replacing YOUR_COM_PORT with your com port, to restore the Oak to factory setup so that the Serial upload will work and the Oak will boot to it.
Use "Upload Using" "Serial" in the Arduino tools menu.
Upload from the IDE and test

## Bounty

$100 cash or $200 credit or 20 Oaks

If you are sure you can do this, and able to do this quickly - please feel free to respond to this issue and say that you are working on it and when you will complete it, so that others don't waste their time on it. If you are just entertaining the idea/unsure if you can do it, please don't "claim" it until you are sure.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. The winner will be the first pull request that submits fully working code meeting the above requirements and following good coding practices, based on the timestamp of the pull request. Bounty will be awarded (or in the case of Oaks, sent) within 48 hours of confirming winner. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

emcniece, 2016-02-12:
I think this one is for me! It's definitely in my neck of the tech woods - will crack it open over the weekend.

kh90909, 2016-03-28:
Okay, I've gotten far along enough with this that I'm confident I can finish it in a reasonable amount of time, so I'm going to claim this issue. @emcniece I hope you don't mind, it doesn't look like you're working on it any more.

So far, I've found and fixed a few bugs in the OakCore cloud serial code, added rate limiting, and I've got a prototype jQuery-based terminal app successfully logging in via ParticleJS, populating the dropdown lists (devices, variables, and functions), and displaying the output of `Particle.print()` running in a loop on my Oak. What remains is to flesh out the app to meet the other requirements (send commands, data to the device, add a settings dialog, etc.)

I hope to get something up for testing in a few days, depending on how busy I am at work this week.

digistump, 2016-03-28:
@kh90909 Great to hear, thanks for working on this and thanks for the update - I look forward to seeing it!

emcniece, 2016-03-29:
@kh90909 feel free to dig in! I'm basically at the same point (basic UI, device list) but it sounds like you have made it further into the core. 

If you would like to collaborate, I'm open... but I have definitely not made "timely" progress :D

kh90909, 2016-03-31:
@emcniece I'm happy to collaborate.

I've uploaded the cloud serial bugfixes to my [fork of OakCore](https://github.com/kh90909/OakCore/tree/cloud-serial-fixes), and put my work-in-progress html/js terminal app in it's own repo [here](https://github.com/kh90909/OakTerm). You should be able to run it via the rawgit link in the readme. Hack away!

kh90909, 2016-03-31:
@digistump I'm running into some difficulty sending commands and data *to* a specific Oak. Right now, `oak/device/stdin` and `oak/device/reset` events go to all of your Oaks. This could be useful on occasion, but definitely not always. I don't see any ParticleJS or Cloud API call to send an event to a specific device id. Am I missing something obvious? It looks to me like the Particle Cloud doesn't support sending events to specific device ids.

If I'm not missing the obvious, then I think a good way to emulate this functionality is to create stdin and reset events for each device, i.e.

```
oak/device/stdin/:deviceid
oak/device/reset/:deviceid
```

and have each Oak listen for the events matching its own deviceid.

On a separate note, I've submitted a pull request to OakCore with my cloud serial fixes. There's also one to OakSystem as I found a typo that affects the sending of some stderr messages.

digistump, 2016-04-01:
@kh90909 thanks for all your work on this - pull requests merged!

I've contact particle about the sending to a single device issue (and confirmed that I don't see a way to either, but Particle has similar events, so they must have a trick)  - asking if its possible, and if not, how they make their similar functionality work - if they don't provide a solution then what you suggest sounds like a great work around (feel free to use that in the meantime, and even in code submitted)

kh90909, 2016-04-01:
@digistump no problem, I'm glad I can contribute. @emcniece is doing great work too refactoring my kludged together html/css/js into a polished app.

Okay, I'll try out my `/oak/device/stdin/:deviceid` workaround in the meantime. It shouldn't require many changes to implement.

kh90909, 2016-06-09:
A beta version of the cloud serial interface (aka OakTerm) is now available: https://github.com/kh90909/OakTerm

Please report bugs at https://github.com/kh90909/OakTerm/issues. I've also created a forum thread for help and discussion: https://digistump.com/board/index.php/topic,2302.0.html