digistump/OakCore #31: BOUNTY: Increase router compatibility

digistump, 2016-02-10:
**Skills Required:** Intermediate at C/C++ and ESP8266 knowledge
**Difficulty:** Unknown

## Challenges/Thoughts:

It seems the Oak is having trouble with several routers (with 12,000+ oaks in the wild it is hard to know if its 1% or 10% of routers, but it seems to be the most common issue). The Oak uses the ESP8266 Arduino Core/WiFi library (which uses the ESP8266 SDK and lwip as a base) as its code base for WiFi functionality including connecting to the Particle servers and the update servers.

If you have or can find a router/AP that the Oak doesn't work with, and then get it to work through code changes, we'll pay you this bounty, we'll pay this bounty for every unique fix submitted (meaning if a fix already submitted doesn't fix it for that router, and your fix does, we'll pay it). You must submit the router name, version,and any relevant settings so that we can purchase the router and confirm the fix if we choose too. 

## To test:

Install Python 2.7 and ensure it is in your path
Install the latest Oak core via board manager, download this repository and copy over the files found at APP_DATA/packages/digistump/hardware/oak/XXX where APP_DATA is where your system keeps the Arduino data and XXX is the current Oak core version.
In cores/oak/OakParticle/particle_globals.h uncomment #define OAK_SYSTEM_ROM_4F616B 82 - this tells it to compile the OakSystem sketch as a system image not a user sketch
Grab the latest OakSystem sketch from https://github.com/digistump/OakSystem
Make changes to the core and/or sketch to achieve the above task.
Connect your Oak using a 3.3v USB to Uart adapter, connect Pin2 to GND before powering on to enter Serial bootloader mode (after an upload the code with automatically run and you'll have to power cycle it to go back to bootloader mode. Disconnect Pin2 to GND to have it run the code on next power up)
ONCE: Grab OakRestore (https://github.com/digistump/OakRestore) and run python esptool.py --baud 115200 --port YOUR_COM_PORT write_flash -fs 32m 0x1000 blank.bin 0x2000 oaksetup_restore.bin 0x0081000 oakupdate_restore.bin 0x101000 blank.bin 0x102000 blank.bin 0x202000 blank.bin replacing YOUR_COM_PORT with your com port, to restore the Oak to factory setup so that the Serial upload will work and the Oak will boot to it.
Use "Upload Using" "Serial" in the Arduino tools menu.
Upload from the IDE and test

## Bounty

$50 cash or $100 credit or 10 Oaks (more at our discretion if the fix turns out to fix many types of routers)

If you are sure you can do this, and able to do this quickly - please feel free to respond to this issue and say that you are working on it and when you will complete it, so that others don't waste their time on it. If you are just entertaining the idea/unsure if you can do it, please don't "claim" it until you are sure.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. The winner will be the first pull request that submits fully working code meeting the above requirements and following good coding practices, based on the timestamp of the pull request. Bounty will be awarded (or in the case of Oaks, sent) within 48 hours of confirming winner. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

DarkLotus, 2016-02-14:
Via Oak Support forums two people reporting they cant get the Oak to connect to their WIFI and mentioning Router models.

> My WiFi configuration is an Asus RT-N66U in WPA2 mode.

> My access point is an Apple Airport Extreme (the new tall one, not the old flat one)

dale3h, 2016-02-19:
I can also confirm that I've had issues with RT-N66U w/ Asuswrt-Merlin firmware ([WPA2-Personal](http://imgur.com/edD6kjV)) at a distance of approximately 15m. I will try again on a 2nd Oak at a closer range and report back.

However, using an iPhone 6 Plus as a hotspot at a range of <1m worked very well.

digistump, 2016-02-23:
@dale3h Thanks for the detailed info!

I was finally able to replicate the update issues reliably - specifically they seem to involve fast connections allowing the device to download data so fast that each loop to write the data to the Oak's memory goes so fast that it outpaces the Oaks ability to erase the next block of memory before writing it. (This bug seems to possibly have some deep roots in the chips internal libraries).

I was not able to reliably replicate this previously because, being in a rural area, my internet connection is slower than many (1.5mbps on a good day) and all of my local testing was done with extensive debugging output turned on, which slowed down the loops enough to prevent this from showing up. Stripping my debug output down significantly and ensuring no output occurred during the write/erase loops, while also hosting it locally within my network allowed me to reproduce the issue 100% of the time.

My fix is to have the web server send only one sector worth of data every 50 milliseconds. This results in a total time of at least 150ms per sector, which seems to fix this issue. I've tested it locally and remotely with 100% success.

This fix is live and you can test it by retrying the update (no changes to the Oak or Config App needed).

If the update still fails to work then please download the newest OakRestore here (https://github.com/digistump/OakRestore) and follow the instructions to provide the debugging output so we can look into it further, as this very well could be one issue of several.

(Sorry that's very long winded as I'm copying the text from a Kickstarter update I'm preparing, but seems worth posting here)

dale3h, 2016-02-23:
I tried the update again with the Oak that failed to update many times the other day.

Router: Asus RT-AC68W
Distance: ~12m
Power: MacBook Pro USB Port

Attempt 1: Failed, but flashed rapidly
Attempt 2: Without power cycling, I tried again. It failed, WITHOUT rapid LED flashes.
Attempt 3: Power cycled, saw rapid LED flashes, 1 blink per second, more rapid LED flashes, and back to 1 blink per second.
Attempt 4: Power cycled, LED went blank for about 10sec while trying to update, then back to 1 blink/sec.

I will attempt OakRestore and try again.

pfeerick, 2016-02-23:
Same here. Have just tried with a virgin Oak, and got happy download blinks for about 25-30 seconds and then a failure message when SoftAP reconnected. If I get time tomorrow, I will go through the OakRestore process and post debug logs here.

My router is a Billion 7800, which I believe also uses a Broadcom chipset.  Have tested  2 & 6 metres away from the router (with two working Oaks having no issues at 2m or less proximity).

starwolf73, 2016-02-26:
In case you guys didn't see, I post my experience on upgrading my OAKS and did find some trick that were fine for me. Let me know if it work for you guys. 

Erik, Did you had a chance to look deeper at what I post?

digistump, 2016-03-09:
Continued in #54

closed 2016-03-09 by digistump