digistump/OakCore #35: BOUNTY: Manage this open source project for a month (or more)

digistump, 2016-02-11:
**Skills Required:** ESP8266, Arduino IDE, Particle, C/C++ (or learn them as you go)
**Difficulty:** Easy but potentially time consuming

## Challenges/Thoughts:

I realize this is pretty strange, and I'm sure some will be offended by the very posting of this. Please realize this is NOT in any way an indication that we don't remain fully committed with all our available to time to this project, it is simply an idea in case someone is interested in managing the open source core and config tool side of this project, which would increase our focus on other aspects of the project (new tools, documentation, hardware, dashboard, etc). If no one is interested/the right fit - no problem, we'll keep on balancing priorities as best we can and release often so more and more people can use it - but if someone wants to jump into this more for fun then profit, then that would be awesome for everyone.

Are you a student, Oak/ESP8266 enthusiast/or just having fun working on this project and want to do more? 

Expectations:

- Handle Pull Requests, Issue reports, and minor bugs
- Answer questions on this issue tracker and on the message board (not expected to answer everything, but to be active and helpful)
- Create new firmware releases, test them, update Arduino 3rd party json file, when needed compile tools for all platforms supported.
- Improve SoftAP config app for smoother workflow in both success and failure.  

## Bounty

20 Oaks will be provided to any person selected to do this, to use for testing. 

Please tell us how much you think this is worth (one sum, or milestone/hourly/etc - we're open to ideas) given that we expect it to take sporadic involvement/time over the next month. Payment can be in cash or credit. Please keep the other bounty amounts offered in mind as far as our budget (this is not intended to be an offer/viable employment, just a bonus for being more involved)

If you are interested, let us know here, we'll expect that if you apply you've helped with this project in some way already, so that we know you would be able to handle this.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

nokemono42, 2016-02-12:
I'm interested in helping manage the project. What's the best way to discuss this with you?

eumb, 2016-02-22:
Hi.
I would like to be a part of this project. If my skills are useful I would be happy to join.

Regards.

igrr, 2016-03-03:
@digistump I realise this might not be exactly your plan, and this might not be exactly on topic here, but perhaps you would consider merging your changes into https://github.com/esp8266/Arduino project at some point. We are moving pretty quickly, fixing a lot of issues and helping users, so that would tick at least 3 out of 5 expectations above. Between these two projects we have a lot of shared code and I think it makes sense not to diverge the code bases too much. I realise that you have a commercial product with a separate release cycle, but at the same time i think we can sort out the way releases are managed to suit both projects.
Regards,
Ivan

haakonnessjoen, 2016-03-03:
@igrr I am to sure how well that would work, as digistump is heavily merged with the Particle-cloud system, whereas I would guess the esp8266/Arduino project is more of a "stand alone" type project. So there would be a lot of code pollution merging the two projects?

digistump, 2016-03-09:
@igrr - thanks for commenting, it has been on my todo list for way to long to write something similar over on your project.

First of all, thank you for your countless contributions to the ESP8266 community, and therefore the Oak. I wrote some prototype Arduino cores for the Oak during its development, so I know how difficult given the ESP documentation, quirks, etc that is - and your core is far more complete and feature rich than anything I wrote.

I'd love to see the Oak connected, if not completely merged into your core, I think there is rarely anything better for the community than unifying open source projects that revolve around a common platform. There are certainly a few parts of it that diverge in a major way, but I'm sure with some brainstorming things could be made to be compatible and/or contained within the variant folder for the Oak. Namely this is of course the Particle integration (which is limited to the Oaks because Particle devices must ship with Particle allocated ids), and the bootloader/rom layout (which is based on rboot).

I've created a new issue for continuing this discussion: #55 - I hope we can continue this discussion there.

Thanks!
Erik

digistump, 2016-03-09:
Closing this issue - thank you to the volunteers! Due to the upcoming public release and much more reliable config tool I think we're going just remain at the helm of all the parts for now as handing it off would likely take more time than continuing on. Thanks again!

closed 2016-03-09 by digistump