digistump/OakCore #38: BOUNTY: Make Fritzing Part for Oak

digistump, 2016-02-12:
**Skills Required:** Basic Fritzing/Image Manipulation
**Difficulty:** Easy 

## Challenges/Thoughts:

Use the Oak image from the pinout graphic to make a Fritzing part - please label pins based on an actual Oak as the pinout is somewhat outdated (I'll be updating it next chance I get).

## Bounty

$25 credit or 3 Oaks

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

DeuxVis, 2016-02-12:
Ooops I just posted that pinout graphics on the oak wiki, as I had a hard time finding it I thought I'd help others. I'm going to mention it is outdated there.

3dpinta, 2016-02-15:
This was my first attempt at making a Fritzing part but here's my submission. I'll get working on some tutorials for it as well.
[Digistump Oak (v1.0) Fritzing Part.zip](https://github.com/digistump/OakCore/files/130076/Digistump.Oak.v1.0.Fritzing.Part.zip)

exeng, 2016-02-18:
3dpinta, Nice work on the fritzing part. I too am a fritzing noob. You did all the hard work. I've been working on another form of the Oak graphic that you may want to use. It's a work in progress and I hope to finish it either this evening or end of day tomorrow. You are welcome to use it and take all the credit for the part entry. (See attached zip). The first one will have headers on it. The second (a derivative) will have male pins so that it can be used on the breadboard. The .SVG will give you a good idea of where I am going with it. I think the pins line up better with the breadboard. If and when you pull it in to the part, it will have to have the pins selected again for fritzing.
[OakBoard.zip](https://github.com/digistump/OakCore/files/136656/OakBoard.zip)

3dpinta, 2016-02-18:
I'll take a look at it next week. I'm swamped through the weekend but I will roll it into my part to update it when I get a chance. Thanks!

exeng, 2016-02-18:
3dpinta, OK, based upon your initial hard work, here are two fritzing parts for the Oak with new graphics. One part is Oak with female hearders installed which only connects to a breadboard through wires. The second part is Oak with male headers for all but the Digispark compatibility 3 pin female header. It can be dropped onto the breadboard and connects
[OakFritzingParts.zip](https://github.com/digistump/OakCore/files/137173/OakFritzingParts.zip)
 accordingly. Both parts also now have their internal connections made, which was missing from your part. Take a look and use at will.

Eric, if you have a chance to review this, I labled the pins base upon the silkscreen on the bottom of the board.

3dpinta, I couldn't have done this without your initial work. You deserve the credit. I just wanted to have a better graphic.

3dpinta, 2016-02-19:
Looks good!

exeng, 2016-02-19:
3dpinta, The version with the female headers needs some tweeking. Try to use it for an example sketch and noticed the labels are cutoff. Must have got cropped somehow. I will fix and resend.

exeng, 2016-02-19:
3dpinta, et al, OK this one should be it. Fixed the cropping problem that was hiding labels. Changed the graphics slightly.  The FZPZ files have the same names but extract to a different directory. Due to a Fritzing issue, deleting to old version and does not completely remove the part which prevents importing the new version. I had to find where Fritzing keeps the files and delete them manually. If you haven't already installed the previous files, your import should just work. Happy Fritzing. Let me know if you have or see a problem with this one.
[OakFritzingParts(V1.0).zip](https://github.com/digistump/OakCore/files/138400/OakFritzingParts.V1.0.zip)

digistump, 2016-02-19:
@3dpinta - thanks for making this, I will send a email about the bounty. @exeng - thanks for improving it!

closed 2016-02-19 by digistump

exeng, 2016-02-21:
@digistump , I realize this was ask for or needed but to complete my Fritzing learning exercise I decided to do a Fritzing part for the Level Shifter. It's mostly done but I'm fighting Fritzing trying to get the PCB silkscreen to showup. Fritzing apprears to be a very powerful and useful tool but it can be frustrating trying to understand how to make it happy. I'll post when completed but it may be a few days as I have some other matters that need my attention.

exeng, 2016-02-22:
@digistump Erik here is a simple Fritzing example to show off the Level Shifter breadboard part. There is some sort of artifact just below shifter (not an Oak) that I'm trying to figure out why is there. Don't see it in my SVG file. In any case this is just so you can see the part. I'll release the part once I'm happy it's right.
[LevelShifterExample.zip](https://github.com/digistump/OakCore/files/139951/LevelShifterExample.zip)