digistump/OakCore #39: Create Mini-tutorials for Starter and Ultimate Kits

digistump, 2016-02-12:
**Skills Required:** Basic Fritzing/Image Manipulation/Writing
**Difficulty:** Easy 

## Challenges/Thoughts:

Use the Oak Fritzing part (this can't be done until someone does that - see #38) to put together a simple Fritzing connection graphic of a basic lesson that can be done with the Starter or Ultimate kit.

Provide text for this lesson, including how to do it, what it teaches, and what the components involved are and very briefly how they work/what they do (ex: An LED produces light when powered)

Provide example code for the user to run on their Oak, keeping it simple and well commented.

Provide the image (with a 300+ dpi version linked), text, and example code as a wiki entry at digistump.com/wiki/oak/tutorials (if you are the first make a new list on that page for links to these lessons)

**Lessons that need to be done**
This list will have many more added to it tomorrow:

Blink
Basic switches
Basic LEDs
RGB LED

## Bounty

$25 credit or 2 Oaks per lesson, one person can do as many as they would like.

Cash or credit is your choice. Cash to be paid via Paypal. Credit has no expiration but can only be applied to a single order and does not cover shipping (because that is how our shopping cart works, not because we want to be limiting). Oaks reward includes shipping.

You may credit yourself in the files as well, leaving in tact existing licenses and credits.

Legal Stuff: We will choose a winner at our sole discretion. Cash awards will be made in USD. This is not an offer for hire. All work submitted becomes the property of Digistump LLC to be used at our discretion in compliance with any associated licenses. Void where prohibited by law.

jwhendy, 2016-03-18:
Not sure it counts as a starter or ultimate tutorial per se, but just wrote a guide for connecting to the Oak via serial through an Arduino Uno. I was primarily interested since I didn't have a 3.3v UART but did have an arduino. I figured a lot of others could be in the same boat so after successfully using a chip-less Uno to apply OakRestore, I wrote it up.

Links:

- Got some helpful input and initial feedback from @DeuxVis on #56 
- The [tutorial](http://digistump.com/wiki/oak/tutorials/serial_through_arduino) on the wiki

DJFliX, 2016-03-19:
I'm working on a button tutorial, both using digitalRead and interrupts. I have the text, but did not include the Fritzing yet, as I'm currently out of time. Sorry about that. Will be working on the rest tomorrow:

- [ ] Use the Oak Fritzing part (to get it see #38) to put together a simple Fritzing connection graphic of a basic lesson that can be done with the Starter or Ultimate kit.
- [X] Provide text for this lesson, including how to do it, what it teaches, and what the components involved are and very briefly how they work/what they do (ex: An LED produces light when powered)
- [X] Provide example code for the user to run on their Oak, keeping it simple and well commented.
- [ ] Provide the image (with a 300+ dpi version linked), text, and example code as a wiki entry at digistump.com/wiki/oak/tutorials (if you are the first make a new list on that page for links to these lessons)

jwhendy, 2016-03-19:
Took a stab at the [blink tutorial](http://digistump.com/wiki/oak/tutorials/blink). Feedback welcome. Since there was another tutorial for LEDs, I stuck with the on-board LED. If no one else claims it, I plan to write the following other tutorials (will update if I add more):

- LED basics
- accessing the Oak's data using `Particle.variable()`
- telling the Oak to do something with `Particle.function()`
- using `python` to read/send data with the two above methods

Let me know if there's any feedback on these proposals.

jwhendy, 2016-03-19:
@DJFliX Nice! If you need any help on the Fritzing, I could do the pulldown vs. float one. No pressure to accept... just throwing it out there.

Regarding the pulldown stuff, I used to do this until I [found out](https://www.arduino.cc/en/Reference/PinMode) one could do `pinMode(n, INPUT_PULLUP)`. I use this exclusively so I don't have to use physical resistors. It inverts all of your code, and you need to wire to the pin and ground vs. the pin and Vcc. If you've got a bunch of buttons/sensors that need to be anchored, it eliminates a lot of resistors! I could write up a section about this; otherwise, that was the feedback that came to mind when reading your tutorial.

jwhendy, 2016-03-20:
@DJFliX Had to make some Fritzing's myself and made a floating vs. pulldown button example for you. Your choice to keep or re-do. Hopefully I got close to what you were looking for there.

DJFliX, 2016-03-20:
@jwhendy Thank you very much! I don't have any experience in Fritzing at all, just got to creating an image of the set up. Thanks a lot for adding the pullup/pulldown images! The internal pullup/pulldown functionality would be awesome, but because this is a breadboard/tinkering type of lesson I thought it would be better to explain the pullup/down mechanism with physical parts first. It's a "hello button" exercise after all :). 

Also: after viewing your images on the buttons lesson, I realised that the female Oak is much more sensible to use in the example. Going to redo my fritzing now :P

DJFliX, 2016-03-20:
RE: http://digistump.com/wiki/oak/tutorials/button
@digistump Erik, I've made some changes in the text and uploaded both instructions to build the circuit and the Fritzing image of the circuit. @jwhendy kindly provided examples of a pull-down and a direct connection of a button to the Oak. Could you (swiftly) have a look at the page and tell me if there's anything missing, or anything you'd like me to add?

jwhendy, 2016-03-20:
@DJFliX: no problem at all. With just a couple of components it went fast :) Nice work, really liked your layout/format.

exeng, 2016-03-20:
I will claim the Temperature Sensing and the 1-Wire Bus tutorial.  Draft is ready. Just need to transfer to DokuWiki and do some final tests.

exeng, 2016-03-21:
Here are the images for the Temperature Sensing tutorial.
[Oak-Temp-Sensing-1-Wire.zip](https://github.com/digistump/OakCore/files/181586/Oak-Temp-Sensing-1-Wire.zip)

digistump, 2016-03-21:
@DJFliX - that is a great little tutorial and laid out perfectly, thank you! Please email us to claim the bounty. 150R is fine, using the external LED was perfect and a great way to build on the LEDs tutorial that will be before that.

@jwhendy - thanks for doing those two tutorials, they look good. Could I ask you to format them to follow the sections and formatting recommended here: http://digistump.com/wiki/oak/tutorials/lesson_template, and to also break the First Blink tutorial down into some small steps as far as the hookup, and maybe add a diagram with the LED labeled (or some other way of making it super basic)? The First Blink tutorial is really essential that it uses super super basic steps since for some it may be the first time they've ever touched a development board. The photocell diagram and gifs on both are awesome! Please email us (support at digistump dot com) whenever you are ready to claim your bounties.

@exeng - as noted (but not very clearly) in the description please upload the images directly into the wiki for these. Your tutorial is very comprehensive and has lots of great advanced detail, but it isn't very beginner friendly in its language, layout, and its lack of small steps with diagrams. I've moved it to a different section (as part of the temp shield tutorial: http://digistump.com/wiki/digispark/tutorials/temp#programming) and left the dead link for the starter kit lesson, feel free to tackle that using the same info you have put together but greatly simplified with diagrams and following the layout recommended (and you can collect the bounty for that as well).
If you'll upload the images into it I'm still happy to pay you the bounty on this of course - it really is a great advanced tutorial and most valuable to have! 

@ Any one submitting a tutorial, please add your name or however you'd like to be credited to your wiki page.

jwhendy, 2016-03-21:
@digistump Will do, and sorry about that! I saw the note on formatting initially but it just took me to the wiki syntax. Missed the template above... I'm in the process of writing a more advanced LED tutorial, want to write an intro to `Particle.function()`, and one on pushing something to the oak with `python` to complement the other one I did.

Any other feedback is welcome... I'm reasonably new to microcontrollers (maybe 2 years with an Arduino to date), so I was really shooting in the dark regarding how basic/advanced to target!

digistump, 2016-03-21:
@jwhendy No problem, I apologize if the formatting expectations weren't clear.

The most basic steps are the goal of these tutorials since they are for the "Starter Kit" which we found generally sold to people with zero or nearly zero electronics experience (with some exceptions of course)

exeng, 2016-03-22:
Erik, sorry for creating the tutorial using the wrong template. With regard to the content being moved to the shield tutorial, the images do not include the temp shield but rather are fritzing diagrams of the Oak and the required discrete parts (i.e. pull-up resistor and DS18B20). The only generic image that could be applied to the temp shield tutorial is the first one as it is a screen shot of the particle.io log. Let me take a second look at the template and if I decide to claim the kit tutorial again, I will post my intent here. Should have read the instructions carefully. Got in a rush since I had the draft ready for awhile. And yes, there was that voice that said this is too much for a simple tutorial.

exeng, 2016-03-22:
OK, I will claim and take another stab at the Temperature Sensing and the 1-Wire Bus tutorial for the Kit lessons.

exeng, 2016-03-22:
OK, completed the Oak kits Temperature Sensing 1-Wire lesson. Images have been included in the Wiki as instructed.

Do not use the Oak-Temp-Sensing-1-Wire.zip images that I previously sent for any tutorials as I discovered that my connections to the DS18B20 were backwards in the files sent.

Let me know if you need anything else.

exeng, 2016-03-22:
Erik, I will fix the images for the advanced example in the Temperature Sensor (1-wire) Shield tutorial and try and get those in tomorrow. Oak Kit is done as previously mentioned.

exeng, 2016-03-22:
@digistump 
Erik, There is a problem with the thumbnail for the circuit image in the Kit Temperature Sensing lesson. During editing I updated the image which is shown correctly when selected to link to the original but the thumbnail that is shown in the Wiki page is the old one. Can't figure out how to fix.

exeng, 2016-03-22:
@digistump  Erik,
OK, The image is correct now. Done! I hope.

jwhendy, 2016-03-25:
I'm going to call it good on my contribution toward this effort; shot you an email last night @digistump .

nog3, 2016-03-26:
I'm working on tutorials for:
- Making noise with the buzzer
- Tilt Switch
- Using a Servo
- Ultrasonic Ranging

and possibly:
- Soil Moisture Sensors
- Powering the Oak with a 9v Battery and Breadboard Power Adapter

But the fritzing part with the pins isn't wide enough to fit in the 400 pin breadboard and is slightly offset from the pins it claims as shown in this image. 
[image]

It does however attach the right pins, so I can work with that but I'm not 100% comfortable with how weird it looks. As a workaround I can do most of the above with the female header populated Oak and the male to female cables but that's less tidy than the breadboard.

@digistump what's your thoughts on this?

jwhendy, 2016-03-26:
The board should probably be modified, however in the meantime you *could* cheat and manually adjust the `x` position of the board. I just tried it and ended up having to move it 0.03 from where it snapped to position. Looks reasonable:

[image]

It was origionally at `x = -0.459`; ended up looking right at `-0.429`.

exeng, 2016-03-26:
Someone must have created a derivative of the original Fritzing part I submitted with female headers which do not line up with a breadboard. Should be and easy fix for whoever created the pin version.

exeng, 2016-03-26:
BTW, where do these parts live? I have the original Oak (female header version) and just posted to #39 a Level Shifter Shield part but where is this new Oak pin version?

nog3, 2016-03-26:
The pin version I'm using is OakFritzingParts(V1.0).zip/Digistump Oak (v1.0-HeaderPins).fzpz from https://github.com/digistump/OakCore/issues/38#issuecomment-186315989

exeng, 2016-03-26:
Man, I must be losing it. That's the part(s) zip I submitted. I forgot that I created both versions. Checked my parts in Fritzing and the pin version snapped right in and lines up. Don't know why it wouldn't for you.

nog3, 2016-03-26:
@exeng As it turns out, snap to grid will place the part according to its entire size, not just the pins you'd expect it to snap with. After turning off snap to grid it works fine.

digistump, 2016-03-26:
If anyone in this thread it waiting for a bounty and hasn't gotten it -
please email me at support@digistump.com - I think I've responded to all
the bounty claims I've gotten, but I want to make sure I didn't miss any.

Anyone adding new tutorials, once they are up please post back here, and
then email me to claim.

nog3, 2016-03-27:
Hi @digistump http://digistump.com/wiki/oak/tutorials/buzzer is now up and should be right to go.

I'll post when the rest are up and send in a bounty claim when you're happy with the content for them :)

exeng, 2016-04-19:
@jwhendy, Looks like someone in the forum found an error in you RBG LED tutorial. See this message...
http://digistump.com/board/index.php/topic,2211.msg10333/topicseen.html#msg10333

jwhendy, 2016-04-20:
@exeng Bah. Thanks for the heads up! I think I recall seeing that locally, making a mental note, and then forgetting to change it on the wiki. Fixed.

nog3, 2016-05-09:
Tilt Switch has been up at http://digistump.com/wiki/oak/tutorials/rollingball for a while, happy to fix any feedback.

brohan, 2016-07-30:
Hi @digistump 

I just completed the Ultrasonic tutorial (http://digistump.com/wiki/oak/tutorials/ultrasonic).
 
Everyone, this is my first tutorial, I would appreciate your reviewing it for errors, and giving me feedback. It appeared to me that one either needed a level shifter (not included in the starter kit) to properly run the 5v HC-SR04, or a different ultrasonic sensor (the route I went).

How about a spot for a level shifting tutorial?

brohan, 2016-08-05:
I added anther UltraSonic tutorial to the main Oak Wiki page using the included HC-SR04, with a level shifter. It can be found here:
http://digistump.com/wiki/oak/tutorials/ultrasonic_2

brohan, 2016-08-12:
@digistump Completed the PIR tutorial. No additional coding or Oak concepts, but focused on investigating sensor datasheets more thoroughly to see an additional uses or applications that what's advertised, and encouraged the reader to post their project on the Oak forums.

http://digistump.com/wiki/oak/tutorials/pir

brohan, 2016-08-13:
@digistump I updated the PIR tutorial to something I am happier with. I included a new concept expanding the Particle.publish to using that info via ifttt.com.

brohan, 2016-08-14:
@digistump I am claiming the Bounty on the powered breadboard tutorial. In this tutorial I incorporated a Deep Sleep function as a Concept. Please review and give any feedback for improvement and corrections.

[http://digistump.com/wiki/oak/tutorials/breadboard_power](url)

rkinnett, 2017-01-15:
@digistump I added a stepper motor lesson.  I hope it's not too much detail...

[https://digistump.com/wiki/oak/tutorials/stepper_motor](https://digistump.com/wiki/oak/tutorials/stepper_motor)

I hope someone finds this useful!