digistump/OakCore #44: Possible wifi SoftAP gremlin?

pfeerick, 2016-02-15:
I'm wondering if I've come across a part of the bug that people seem to be having with routers and SoftAP setup - my SSID has a '(', ')' and some ' ' (spaces) as part of the name, and I couldn't get the Oak to connect via it after several attempts. However it immediately connected through my Ravpower Filehub as a AP instead, which had an SSID of 'Filehub-4CU2', so no spaces and only symbol was a hyphen. Would like to hear other peoples thoughs in case it is just a gremlin on my end - as otherwise I won't be able to use the Oaks with this AP until either the bug is fixed, or will have to use a guest SSID just for the Oaks.

I created a secondary guest SSID on the same router that I was having issue with, creatively named OakSSID, and a second Oak (or lemming?) I connected without any issues. After a kick or two up the bum to get it to update, it give me the happy 0.5sec blinks, and SoftAP is trying to connect to it again, and hopefully I can connect this one to Particle also, but without having to have a secondary AP for to connect through...

pfeerick, 2016-02-15:
Hm... it appears lemming #2 is not connecting to Particle... although it was able to do the firmware update. Looks like I'm stuck using a secondary AP until this is resolved. I'll update tomorrow, as I'll leave it connected overnight to see if anything shakes loose... stupid thing is... it is connected to the network ok, as I can see it's been allocated an IP and is responding to pings... just isn't connecting/registering itself on particle... :( And upon closer inspection of my routers logs, it looks like the other Oak connected to the original SSID, but also didn't register itself on particle, hence the workaround of a second AP... so maybe that is the heart of the problem?

pfeerick, 2016-02-15:
Hm... pulled the secondary AP and tricked the first oak which had been registered with particle into rebooting into SoftAP mode, as I couldn't get the Pin0 to GND trick to work for me... Told it to use the OakSSID and it connected, and is active in particle now... very interesting indeed.

pfeerick, 2016-02-15:
And another update on the second module - after it was registered on particle, I tried to program it via the OakSSID AP, and it didn't seem to take... was doing it's three blink with random blips in between and nothing happening after 3-4 minutes of that performance. So I tricked it back into AP mode (a power cycle was all it needed that time?), and set it back to the filehub AP, and was able to program the blink sketch. Then shutdown the filehub AP, waited for the SoftAP recovery to kick in, and then connected it to my main AP (with the weird naming). It accepted it that time, and particle acknowledged it was online at last. I then uploaded a second blinky sketchm and it is working fine. 

So it appears (for me at least), the issue is getting the first update, and particle registration and the first upload to happen, I have to use the filehub AP to get things going. Once that first program is one, it seems to be happy to connect and program over whatever it is told to.

digistump, 2016-02-23:
@pfeerick - can you retry the initial update using your main AP? I think this might have been unrelated to the AP name and just because the filehub AP was a bit slower or something - see for more: https://github.com/digistump/OakCore/issues/31#issuecomment-187462037

pfeerick, 2016-02-23:
I agree... initial update issue may be speed related. Will respond with my results in other thread.

closed 2016-03-23 by digistump