digistump/OakCore #47: Oak Update Debug Logs

digistump, 2016-02-23:
Please paste you Oak update debug logs here - or email them to support@digistump.com if you do not have a github account to comment here.

Reference:

ancientmidi, 2016-02-23:
rlØrß#ânÀàìp<ßì8Çäpònnä;ònÜäbç$rrpònàÃÜàÇÀlbÀnâãnÀäìbÀÄ¾~ònîÄÁl`Ü#nÀrnrß;ÄÀ?Ç;pònàÃÜàrâpp<ÇlÄbÀÄ>~ònîÄðrÜ#nÀrnrß;ÄÀ?àrrÛnàrr~Ûà¹nònnäÄÜÇlbÀü|ÿÛÛÁbà$``
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0

rlrß#ânÀàìp<ßì8ßÇÜäpònnäÄ;ònÄÜäbà$	rrpònàÃÜàÀlbÀnâãnÀäìbÀÄ¾~ònîÄÁl`Ü#nÀrnrß;À?Ç;pònàÃÜàrâpp<ÄÇlbÄ>~ònîÄðrÜ#nÀrnr;ÀàrrÛnàrr~Ûà¹nònnäÄÜÇlbÀü|ÿÛÛÁbà$``
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
[... 280 lines cut ...]
OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0

digistump, 2016-02-23:
Thank you  - I was able to duplicate this and I'm working on it now
Will update here when I think I have something

digistump, 2016-02-23:
@ancientmidi If you can retest I'd be interested to know if it is working for you now - note you will need to re-download oakupdate_restore.bin from the repository

ancientmidi, 2016-02-23:
Not noticing a difference but I'm going to re-clone OakRestore to make sure I didn't use the wrong oakupdate_restore.bin

ancientmidi, 2016-02-23:
Will confirm with another in a bit.

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+
 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0

rlØrß#ânÀàìp<ìxÇäpònnäÄ;ònÄÜäb#l``pònàÜàÀlbÀnâãnÀäìbÀÄ>~ònîÄÁl`#nrnr;À;pònàÃÜàrÇâÜpp<ÇlbÀÄ>~ònîÄàr#nÀrnrß;ÄÀ?àrrÛnÀàrr~Ûà9nònnäÄÜÇlbÀü|ÿÁbà$``
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0

rlrcânÀàìp<ßì8ßÇÜäpònnäÄ;ònÄÜäbà$	rrpònàÜàÇÀlbÀnâãnÀäìbÀÄ>~ònîÄÁl`Ü#nÀrnrß;ÄÀ?Ç;pònàÃÜàrâppÜ<ÄÇlÄbÀÄ¾~ònîÄðrÜ#nÀrnr;ÀàrrÛnÀàrr~9nònnäÄÜÇlbü|ÿÛÛÁbà$``
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+UPDATE OK - BOOT TO ROM
39

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,8

ancientmidi, 2016-02-23:
Sorry if that didn't sound like good news -- let me rephrase.  The one I called "problem_child" activated after a few tries. :)

ancientmidi, 2016-02-23:
Second Oak, not so good.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0

rlØr#ânÀàìp<ßìxÇäpònnäÄ;ònÜäbcl``pònàÃÜàÇÀlbnâãnäìbÄ>~ònîÄÁl`Ü#nÀrnrß;À;pònàÜàrâÜppÜ<ÄÇlÄbÀÄ>~ònîÄðr#nÀrnrß;ÀàrrÛnÀàrr~Ûà9nònnäÜlbü|ÿÛÛÁbà$``
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
[... 278 lines cut ...]
START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0



dale3h, 2016-02-23:
Using the same Oak I've had fail 20+ times. I used OakRestore to restore it (took me a while to figure out...notes below)

#### OakRestore Notes
* I am not a serial expert, so this may just be my lack of serial knowledge: Make sure you do not try to connect the serial terminal before trying to flash, otherwise it will result in an "access denied" error.
* I had to remove PIN2<->GND after flash, otherwise when it reset it would go back into flashing mode.
* I had to power cycle the Oak after it wrote the WiFi settings to get it to start the update

**TTL Cable:** [Adafruit USB to TTL Serial Cable](https://www.adafruit.com/products/954)
**OS:** Windows 7 Ultimate 64-Bit
**Config Tool:** Saved to disk, ran locally in Chrome Version 48.0.2564.116 m

**Router:** Asus RT-N66U
**Distance:** ~15m
**Power:** Belkin Powered USB Hub -> LeGioN Meter -> Oak

	 ets Jan  8 2013,rst cause:2, boot mode:(1,0)

	 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

	wdt reset
	rll|là|
	lì
	b|ìrb
	bònnlnnâì
	bp
	lblrlpònà
	l
	b
	nâ|ì
	bònnîll`nn
	l`nrn
	rl`pònà
	râà
	b
	nâ|ìlp
	ònnî
	l`nn
	l`nrn
	l`rn
	l`rl~rn`~ònplbü|â
	brlrlSTART UPDATE ROM
	WIFI
	WIFI CONNECT
	GO TO UPDATE
	START UPDATE
	HOST LOOKUP OK
	PARSING HTTP HEADER
	HTTP/1.1 200 OK
	FILE LENGTH: 778096
	START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
	./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
	UPDATE FAILED

	 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - N,BU,0

pfeerick, 2016-02-24:
As with Dale, I was tricked by there not being a mention of a 'you can remove the P2 -> GND link after the restore firmware has been successfully downloaded!'... unsuprisingly not much happened until I did... lol

**Attempt 1: Power cycle after OakRestore**
```
START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK

FILE LENGTH: 778096

START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0
```

**Attempt 2: Didn't bother power cycling manually, looks like SoftAP triggered a reboot itself**
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK

FILE LENGTH: 778096

START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0
```

haakonnessjoen, 2016-02-24:
So I see a lot of "SOCKET READ TIMEOUT". Would it be feasible to handle this with a restart of the download, or even a restart at the bytecount it has downloaded, instead of rebooting? This way, if someone have a very crappy internet connection, eventually, it will have downloaded the whole file. Instead of downloading the start 20 of the image 50 times.

Maybe it increases the size of the bootloader too much? But it should not really, if this is infact HTTP requests, you could just add a Byte-range: header?

Just my two cents

pfeerick, 2016-02-24:
@digistump Is having access to more off-site Oaks of any benefit to you? If you are able to access other Oaks by having their deviceID, I am more than happy to leave the two Oaks that are talking to Particle online as remote programmable targets. At the moment they are simply running - there is nothing connected to them. They just publish a "loop()" event every 20 seconds (and blink their LEDs) to show they're still alive and kicking.

daggeisler, 2016-02-24:
Dear Erik get the same error ...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK

FILE LENGTH: 778096

START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK

FILE LENGTH: 778096

START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0

daggeisler, 2016-02-24:
[eth_trace.zip](https://github.com/digistump/OakCore/files/145458/eth_trace.zip)
Hello Erik attached an ETH Trace nothing changed it looks like the server and the oak getting interupted because of retransmits...

dale3h, 2016-02-25:
I was able to get my problem-child Oak to update using the latest beta and my iPhone in hotspot mode.

tifarley, 2016-03-01:
Any updates?? How about giving us another oakupdate_restore.bin that will give slightly more meaningful debug information instead of just socket timeout?

dale3h, 2016-03-02:
Still no dice trying to update my "problem Oak" from restore to 0.9.5:

**Attempt 1:** failed with "SOCKET READ TIMEOUT"

	ets Jan  8 2013,rst cause:2, boot mode:(3,0)

	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - N,BP,2

	START UPDATE ROM
	WIFI
	WIFI CONNECT
	GO TO UPDATE
	START UPDATE
	HOST LOOKUP OK
	PARSING HTTP HEADER
	HTTP/1.1 200 OK
	FILE LENGTH: 778096
	START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
	./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
	UPDATE FAILED

	 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - N,BU,0

**Attempt 2:** power cycled, failed with "SOCKET READ TIMEOUT"

	ets Jan  8 2013,rst cause:2, boot mode:(3,0)

	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - N,BP,2

	START UPDATE ROM
	WIFI
	WIFI CONNECT
	GO TO UPDATE
	START UPDATE
	HOST LOOKUP OK
	PARSING HTTP HEADER
	HTTP/1.1 200 OK
	FILE LENGTH: 778096
	START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
	./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
	UPDATE FAILED

	 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - N,BU,0

**Attempt 3:** did NOT power cycle, failed with unknown error

	ets Jan  8 2013,rst cause:2, boot mode:(3,0)

	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - N,BP,2

	START UPDATE ROM
	WIFI
	WIFI CONNECT
	GO TO UPDATE
	START UPDATE
	HOST LOOKUP OK
	PARSING HTTP HEADER
	HTTP/1.1 200 OK
	FILE LENGTH: 778096
	START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
	./+./+
	 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

	wdt reset
	load 0x40100000, len 3632, room 16
	tail 0
	chksum 0xc0
	load 0x3ffe8000, len 352, room 8
	tail 8
	chksum 0x82
	csum 0x82

	OakBoot v1 - H,BU,0

rei-vilo, 2016-03-04:
Please find the trace of my failed attempts.

* CONNECT P2 - GND
* RESTORE FACTORY FIRMWARE 

* REMOVE P2 - GND
* POWER CYCLE
* 1st ATTEMPT WITH LAN-1

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x40100000, len 3632, room 16
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK
FILE LENGTH: 778096
START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+
 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0
```

>
Your Oak failed to download the update. Please power cycle your Oak, reconnect to its AP, and then click "Retry" to try again with your current settings or "Change Settings" to try a different WiFi network.
>

* RETRY

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0
```

* POWER CYCLE
* 2nd ATTEMPT WITH LAN-2

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - H,BU,0
```

This is an end-less game. Thank you for your help.

pfeerick, 2016-03-07:
Hey @rei-vilo - what you you mean by "too big to download, even at 15Mbps" - by my calculations, at 15Mbps, it would take less than half a second to download 778096 bytes! The bottleneck is how fast the ESP8266 can clear and write the saved data. 

And if you are trying to get an Oak to update so you can use it, trying retrying another 2-3 times (with power cycles) on your LAN-1, as it seemed to be the only one of the two downloading anything. And remember that you do have to connect the oaks ACORN-XXXXX AP again, and hit retry in the SoftAP config webapp, for it to actually retry the download - it doesn't retry automatically or anything.

rei-vilo, 2016-03-07:
I've already tried 6 times with power cycles, to no avail.

DeuxVis, 2016-03-12:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK

FILE LENGTH: 778096

START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0
```

fbrinker, 2016-03-15:
My short story:
I have three devices. Tried to power them by a mobile micro usb loader. All three failed multiple times(x). Connected one to the PC. Failed multiple times(x). Moved one oak with a powerbank a few rooms away (~30m), tried multiple times(x), no chance to update the firmware.

So i soldered the headers to the boards, upgraded on oak1 the firmware via Uart Adapter and afterwards (connected to my pc again, like the previous tries), the firmware download worked like a charm... Connected the second one to the pc again, tried it without firmware upgrade, and that also worked like a charm... well i guess i've found the perfect wifi spot... connected the third one. Didn't work. Upgraded the firmware. Tried it again multiple times(x), always failing. So i will flash the firmware via Uart Adapter to my last Oak now. I hope the normal connection when using them is more stable. :)

```
ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BP,2

START UPDATE ROM
WIFI
WIFI CONNECT
GO TO UPDATE
START UPDATE
HOST LOOKUP OK
PARSING HTTP HEADER
HTTP/1.1 200 OK

FILE LENGTH: 778096

START WRITING UPDATE - NO OUTPUT SHOULD BE EXPECTED FOR UP TO 120 SECONDS
./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+./+.SOCKET READ TIMEOUT
UPDATE FAILED

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 3632, room 16 
tail 0
chksum 0xc0
load 0x3ffe8000, len 352, room 8 
tail 8
chksum 0x82
csum 0x82

OakBoot v1 - N,BU,0
```

(x) multiple times = ~5 to ~10

rei-vilo, 2016-03-15:
The biggest problem is the Oak isn't reliable. I've given up.

fbrinker, 2016-03-15:
@rei-vilo wouldn't say so - the *only difficulty* (and a frustrating start) is to update the firmware to v1. After flashing the last one with my Uart adapter, they work fine... even installing my scripts over the cloud via Arduino IDE -> Particle.io -> My Oak

But yeah... without an Uart adapter... i guess it would have been impossible :(

digistump, 2016-03-23:
Closing given the many ways to troubleshoot this now and logs not revealing anything new at this point:
digistump.com/wiki/oak

closed 2016-03-23 by digistump