digistump/OakCore #51: Oak doesnt connect to particle

Snudi, 2016-02-26:
Hi guys,
yesterday i received my Kickstarter box with 3 Oaks. I still have a particle Account cause i supported the electron project from particle. So i follow the instructions and after some rounds the 1. Oak was updated to System-Version 2. After that he should connect to the particle cloud. But then it stucked. The Oak doesnt connect to the cloud and also wasn´t seen as wifi device until then. So i take the 2. Oak and followed the instructions again. After some tries with my Mobile Phone Hotspot he was upgrade to system-version 2 too. But also this one, doesn´t connect to the particle cloud. Rebootet serveral times since yesterday. This time i had the luck that the oak was still reachable as wifi device. So i could try with my home wifi and two Mobile Phone Hotspots. But it is still the same. He doesn´t connect to the particle Cloud. The 3. Oak ist still in the box. And will be there until No.1 and 2 are running fine. So do you have any ideas what i could do to get them running?

Thanks a lot, Ruediger

digistump, 2016-03-23:
I'm going to close this as troubleshooting wikis and forum posts on how to solve these issues, plus new releases that solve them have all occurred since - digistump.com/wiki/oak or digistump.com/board if you are still having issues with it

closed 2016-03-23 by digistump

Snudi, 2016-03-24:
Sure, sorry for that mess 
Greetings from Germany, 
Ruediger