digistump/OakCore #52: Can't complete 0.9.4 firmware upgrade

ImanolBarba, 2016-02-27:
Hi,

I tried to update from 0.9.3 to 0.9.4

Things I did:
- Update to 0.9.4 in Arduino IDE, also downgraded to 1.6.5 as recommended.
- Git pull the OakRestore repo and flash the new firmware
- Try to configure the Oak using the softAP

Things that happened:
- The firmware flash arduino IDE upgrade went fine
- I started the OakSoftAP (refreshing the cache in the browser) and managed to set the WiFI credentials and also managed to get the Oak to upgrade (the board had a different blinking pattern after the update). After this, the board would no longer set the ACORN-xxyyzz access point so I couldn't set the Particle credentials on it.

I can still put pin 2 to GND and flash the firmware with OakRestore, but I can't complete the upgrade.

pfeerick, 2016-02-28:
So when you go to your [Particle dasbhoard](https://dashboard.particle.io/user/devices), does it show your device there, and is the round circle on the left side of the device ID grey or blue (or pulsing blue)... when the Oak is powered on of course! If so, it is online, and ready to program. You don't log in again to the Oak to set the particle credentials - the sequence is 

1. Load SoftAP on your device (tablet, laptop, phone, whatever)
2. Log into to Particle
3. Switch to the Oak's access point
4. Configure the Oak to access your wifi internet / routher / access point
5. Oak reboots, installs firmware update, and automatically registers with Particle

And as far as I'm aware - 0.9.4 is to be avoided atm due to some bugs found after it was released, hence the rollback warning on the releases page.

ImanolBarba, 2016-02-28:
It wasn't online in the Particle dashboard because the credentials couldn't be set using SoftAP as explained.

I applied the 0.9.4 update before the rollback was announced.

I could checkout a different release and flash it again, I'm just opening the issue to acknowledge the problem and because the 2 other issues relating to the upgrade didn't specifically relate to this issue, which is that the Oak doesn't bring up the AP after the OTA firmware update and because of that, the Particle credentials can't be set

pfeerick, 2016-02-28:
Ok, I got you now. I don't actually think you have to log onto the AP again to get it to set the credentials - that part is only so the Oak can tell SoftAP that the firmware update was successful. I had the same problems with the Particle registration with one of mine, and just restarted the Oak a few times until it finished registration. 

If you want the AP again, start up your Oak with pin 1 to ground to force it to re-enter config mode, and try the wifi config and firmware update again (try pin 6 if it doesn't work - pin is being moved around a bit atm). Otherwise, you might need to resort to a firmware Restore - https://github.com/digistump/OakRestore

ImanolBarba, 2016-02-29:
I tried, and now it sets the AP after the OTA firmware update, but system version is 2, so I guess it downloaded the previous version because of the rollback

I guess we'll have to wait until they fix the issues and re-release to check

pfeerick, 2016-02-29:
System version 2 means it has updated the firmware, but I'm not sure what version it ships with... I seem to recall it being 0. The system version doesn't match up with the Oak core - not sure what the difference there is. So sounds like your problem is that for some reason it isn't registering on Particle, even though it has updated, and is connected to your internet / wifi. That is an Erik problem! You may be able to (while the Oak is in config mode, and you are connected to it's AP) get the device ID by going to http://192.168.0.1/info, and then manually adding that to your Particle account, but I have absolutely no idea how you would do that... only heard that someone else did the same trick.

ImanolBarba, 2016-02-29:
System version 2 is the version of the Beta 3 (AKA 0.9.3), so flashing the Beta 3 from OakRestore would now effectively get you back at Beta 3

Probably because they rolled back

pfeerick, 2016-03-02:
Indeed. It appears from a post from Erik last night the system version is intended to match that last digit of the Oak core version - i.e. 0.9.1 = 1, 0.9.2 = 2, etc..., with 0.9.3 mistakenly left at 2. Just to keep you on your toes! lol

ImanolBarba, 2016-03-02:
Fixed in latest 0.9.5 firmware

closed 2016-03-02 by ImanolBarba