digistump/OakCore #53: HowTo see which version is install and understand the version.

starwolf73, 2016-02-27:
Erik,
To help troubleshoot I'm trying to understand the behind step to take a green OAK and update it to 0.9.x. What is the difference between 0.9.x and the 2 (http://192.168.0.1/system-version) from the OAK? 

Regards,
Eric

pfeerick, 2016-03-02:
Just for completeness, Erik posted this on the forum last night, which answers your question. Can you close the issue since it isn't really an issue? Thanks.

> Shedding some light on system versions:
> 0.9.1-system version 1
> 0.9.2-system version 2
> 0.9.3-system version 2 (mistakenly not incremented)
> 0.9.4-system version 5 (see below)
> 0.9.5-system version 5
> 
> System version jumped to 5 at 0.9.4 because Particle wants to see a version 5 or higher reported to them to enable fast updates on their end. 0.9.4 was quickly rolled back due to major issues with wifi connectivity - since few people installed it I went ahead and kept the version at 5 for 0.9.5 as that allowed me to align the two as well

starwolf73, 2016-03-02:
Well the issue was that 0.9.3 was mistakenly not incremented and that explain the disparity  between the release and system-version we saw and didn't make no sense to us. Just closed the issue.
Thanks.

closed 2016-03-02 by starwolf73