digistump/OakCore #54: BOUNTY: Find a workaround/fix for update issues

digistump, 2016-03-02:
The technical limitations: The firmware shipped on the Oak is solely there to let you configure your wi-fi and get the first update, it, of course - given the Oak does not have a USB interface, cannot be changed except for with this update, so any changes to make this work have to happen on the server side of things.

The technical details: The Oaks preformed very well in our early testing at getting the update, before we sent in the firmware to the factory, while this was tested in a variety of ways the main development setup had the update file hosted on an Apache 2.2 server on Amazon EC2, and was being accessed from our machines over a 3mbps DSL connection. We also tested with it locally on our b/g/n wifi network. Our routers were run stock for testing with b/g/n enabled and auto for channel. - Between this point where we approved the firmware to be burned to the units and when people started to report issues the following changes occurred (or may have occurred): The hardware went from prototype to production, with likely ever so slightly different parts that should not have effected any performance and the update file was moved to an Apache 2.4 server on Ubuntu 14.04 on Digital Ocean (and then to various server setups - see github for more on this). After this updates were still working very well for us, since we didn't think it was an area we needed to worry about I can't say for sure if one ever failed, but they never failed enough to even catch my attention as a possible issue before I started shipping the factory produced units. The final point, and given that the update still works for me more than it doesn't over DSL, but not always over the local network, is why I believe connection speed may be involved.

digistump, 2016-03-02:
Relevant files as mentioned above:

Generic Node File Server with HTTPS: 
[server.zip](https://github.com/digistump/OakCore/files/154426/server.zip)

Attempt at releasing chunks of file more slowly: 
[server_with_delay.zip](https://github.com/digistump/OakCore/files/154427/server_with_delay.zip)

Firmware to load after setting IP/Domain and Fingerprint:

haakonnessjoen, 2016-03-02:
I think maybe file_chunk_write should wait for the write callback to respond, before starting the timer. That way you won't fill up a send buffer if the connection is slow. Which could cause a large burst of data to be sent on a slow connection, I would guess. And this is nitpicking; but getFileSizeInBytes is kind of redundant, since you just read the entire file. So file.length holds your answer to the file size.

epatel, 2016-03-02:
Sounds like the TCP stack in the firmware is broken, i.e. it can't request missing packages. I would hook up wireshark and look carefully of what is going on. If that can be verified, maybe a solution creating a local network with a local server (i.e. in virtualbox or on a mobile device) could minimize package drop.

jldeon, 2016-03-02:
Is there some way to get the source and replicate the build environment for the factory image?  I realize that the final solution must work with the image already flashed, but for debug it would be helpful to see the code on the Oak and make changes.

digistump, 2016-03-02:
The source is the same as attached in the second comment zip file
(oakupdate.zip) - just comment out #DEBUG_SETUP to get exactly what is on a
factory Oak

You can setup the environment to compile it by downloading the following
and unzipping into Documents/Arduino/Hardware/

https://www.dropbox.com/s/dgb4qf1cooz3oba/oak_fallback.zip?dl=0 (Note still
uploading right now, if you go to this link it will tell you when it is
done uploading)

Then select Oak Fallback as the board, Single as the rom config, and hit
upload (assuming you've already followed the steps above to set your wifi
connection and domain/ssl thumbprint)

jldeon, 2016-03-02:
OK, working on getting my environment set up.  

Should I expect any response to the "set" commands from the factory firmware?

digistump, 2016-03-02:
Yes - just added, thanks for asking: The Oak should respond with {"r":0} if after each three lines (set, length, content) are sent - if you get {"r":-1} back then something is wrong with the input.

Also baud should be 115200

digistump, 2016-03-02:
@jldeon - just added some more notes on how to confirm these changes and about timeouts for sending it all, see just under the "set" codeblocks in the top post

jldeon, 2016-03-03:
I've got everything set up and running now, and I'm deep in debug land.  I think I've got a solid lead on part of what's going wrong, but not everything yet.  Will try to keep you posted, assuming someone else doesn't figure it out first :)

DarkLotus, 2016-03-03:
Just started to get things setup to have a play with this, but im in the horrible(joke) situation where my Oak's update fine a good 80%+ of the time. 9/10 in a row this morning. ( The digistump server im talking about )

I've even tried Enabling WPA2 and forcing wifi channel 1 as was advised against. I've also tried hammering my connections up/down stream while updating as well. My only other AP is an Alcatel Pixi 4.5 in AP mode but the Oak just refuses to connect to that full stop.

Im happy to provide remote testing of anyone's WiP solution from a location that seemingly is blessed.

jldeon, 2016-03-03:
I've uncovered one major problem, and I believe the solution is fairly simple.  

This should fix the:

```
HOST LOOKUP OK
NO CLIENT
COULD NOT CONNECT TO UPDATE SERVER
UPDATE FAILED
```

endless loop.

It seems like the network stack is doing something silly, which is causing the server to become confused.  Basically, TCP connections are established based on the client's IP and port, and the server's IP and port.  When most web clients connect to a server, they pick a random port as their client port.  The library on the Oak picks the same one every time (4097).  

If the transmission of the firmware file goes south, the socket is left open on the server.  Most web servers have long timeouts on these sockets, because they expect you to make multiple requests (ie, HTML file, and then a dozen images or JS files or whatever).  

Now you reboot your Oak and try to connect again.  The problem is, the Oak is sending a SYN using the same source IP and port (more than likely, if you're on a home router using NAT).  The server looks at that packet, goes "I have a connection already" and drops it.  

Meanwhile, the server is still waiting for acknowledgement on the last chunk of the firmware file it sent.

We can't fix the silliness on the Oak side, but we can more aggressively close the sockets on the server side.  For Apache, try setting:

```
TimeOut 3
KeepAlive Off
```

In the VirtualHost section (or similar).  This doesn't 100% fix the problem, but it does make it a lot less likely.  [KeepAlive Off](http://httpd.apache.org/docs/2.4/mod/core.html#keepalive) is pretty sensible for the server, since we're not making bulk requests to it.  The [TimeOut parameter](http://httpd.apache.org/docs/2.4/mod/core.html#timeout) gives the client 3 seconds to acknowledge a packet if the send buffer is full (so by that point, you're already behind in terms of transmission).  

On Node.JS, the timeout parameter of the HTTP server object appears to serve a similar purpose.  

Ideally, we'd set something like `TCP_USER_TIMEOUT` on the socket, but that would require either recompiling an existing server, or rolling our own.  I'm not sure that the security and performance impact is worth the trouble, though.  

Another option is setting `/proc/sys/net/ipv4/tcp_retries2` to some lower value.  This would effect everything OS-wide, though, so it's kind of the nuclear option.  I believe the default at this layer is the value 15, which [corresponds to more than 10 minutes](http://linux.die.net/man/7/tcp).  

Before these changes, if I interrupt the download from my local server (pull power while the 123123123 is scrolling), I'll get "NO CLIENT" errors over and over and over again.  With these changes, I can pretty reliably flash my Oak on my local network.

jldeon, 2016-03-03:
Another way we can get stuck in this state "NO CLIENT" is if the web server tries to close the socket, but can't get the Oak to reply to its attempts to do so.  `/proc/sys/net/ipv4/tcp_orphan_retries` controls the number of times that a socket in the FIN-WAIT-1 state (ie, attempting to close) will try to tell the other side that it's closing the socket before just giving up.  The default is 8, and I'd suggest that a much lower number could be used here.

digistump, 2016-03-03:
@jldeon - some great discoveries, thanks! - just wanted to note that we have been setting /proc/sys/net/ipv4/tcp_retries2 to 4 which seems to be a good balance between closing when it shouldn't and allowing frequent retries - this is noted at the top of the server.js file, but this was a shot in the dark - you certainly figured out why this is necessary. In general I don't mind if any of the settings we need to change are OS level - this update server will run isolated on its own cloud/virtual server

jldeon, 2016-03-03:
@digistump  Ah, should have looked at the node.js code :)  I figured you guys weren't going to do anything else on this box, which is why I suggested some of those OS-level changes.  I'm not sure how the kernel does math with that tcp_retries2 value, so I don't know what 4 means in the context of how long the socket will persist.  I'd suggest trying some of the other settings,  as those helped immensely.  

For the record, I'm doing no throttling whatsoever on bandwidth and not having any issues flashing the Oak over and over again.  The update server and the oak are on the same high-speed LAN.

digistump, 2016-03-03:
@jldeon before you implemented those changes were you getting SOCKET READ
TIMEOUT? that is the main issue that seems to crop up for many.

What server are you using to server it? Apache (version, etc?) or Node or
something else?

DarkLotus, 2016-03-03:
@jldeon Do you have any issues receiving updates from the official server though? My local server works fine as well, but I have no issues with the official so cant really debug.

jldeon, 2016-03-03:
@digistump I get the occasional SOCKET READ TIMEOUT, (10% of the time, or so?) in looking at the tcpdump when it occurs, the server is retransmitting the packet but it's not getting to the Oak (at least from the tcpdump on the server, I see no ACK).  I don't know that there's much that can be done about this, though, since we're dealing with wi-fi.  

I'm going to keep digging on that, though, now that I've done what I can with this issue.

jldeon, 2016-03-03:
@DarkLotus Yes, I tried last night and this morning to update with the official server on 2 Oaks, and it failed on every attempt.

DarkLotus, 2016-03-03:
Ah at least you can reproduce :) if you need a vps or anything to test from a remote server let me know.

jldeon, 2016-03-03:
@DarkLotus Thanks for the offer!  I think I've probably got it covered.  I've got a couple of VPSes, credit to AWS, credit to Azure...  probably some other junk if I dug around a bit :)

jldeon, 2016-03-03:
@digistump Totally flaked on your questions.  The box I'm using currently is Ubuntu 14.04.4 LTS, 32-bit, testing with Apache 2.4.7 (latest available in the Ubuntu repos) currently.

digistump, 2016-03-03:
@jldeon - when it failed on every attempt last night against the live
server, was it due to NO CLIENT or SOCKET READ TIMEOUT?

jldeon, 2016-03-03:
@digistump I was running the factory build, so there was no debug output.  I posted what I had to the forums: http://digistump.com/board/index.php/topic,2034.msg9360.html#msg9360

jldeon, 2016-03-03:
I can reliably reproduce the "SOCKET READ TIMEOUT" error by pointing my Oak at my VPS, and I've been digging into it for the last hour or so.

It doesn't look like an actual packet timeout, it looks more like some sort of weird conflict or maybe a race condition?  I see a lot of retransmitted packets on both sides.  

I've got to sleep now, but I'll try and craft an experiment to test this tomorrow if I've got time.

DarkLotus, 2016-03-03:
If anyone is around ive spun up a CentOS 6 server with Apache 2.2 just in case reverting back to Apache 2.2 is the fix.

    set
    43
    {"first-update-domain":"oak.jameskidd.net"}
    set
    90
    {"first-update-fingerprint":"07 99 62 B4 0C 4D 4D 59 22 1F 62 A2 83 04 0B E5 00 94 BE 18"}

At least on my end i can confirm this works at 100% 5/5 times thus far.

Will setup a ubuntu 14.04 apache 2.4 setup next on the same host, and see if i can get some time-outs reliably happening.

DarkLotus, 2016-03-03:
set
    44
    {"first-update-domain":"oak2.jameskidd.net"}
    set
    90
    {"first-update-fingerprint":"07 99 62 B4 0C 4D 4D 59 22 1F 62 A2 83 04 0B E5 00 94 BE 18"}

This one is Ubuntu 14.04 with apache 2.4 All stock as well. Still I cant reproduce socket timeouts reliably, I start to wonder if its router chipset related or something.

DeuxVis, 2016-03-03:
When I see that there are more problems with high speed networks it kinda ring a bell for me : MTU !
Packet fragmentation might not be handled very well by simplified / old IP stacks. I remember it being a problem in the early days of DSL internet access.

So I don't have time to test it myself currently, but trying to lower the server network interface MTU might help there ?

On a different subject, I would be happy to test if any server you guys put out there are an improvement : I am "lucky" enough to not being able of upgrading any of my 3 oaks here at home (Cable internet), and have a 3.3V able USB/TTL adapter available.
Just don't expect lightning speed replys.

DarkLotus, 2016-03-03:
@DeuxVis oak2.jameskidd.net is running with MTU set to 576 if you want to give it a shot.
and oak.jameskidd.net is running apache2.2 on CentOS 6 to try and match Erik's original server

DarkLotus, 2016-03-03:
I have 100% failure rate with the MTU lowered to 576, Socket read timeouts. So Packet size could definitely be a factor in this. Bed time for me, will look at it again tomorrow.

DeuxVis, 2016-03-03:
I will try that later at home. I just successfully upgraded one of my oaks on first try at the local hackerspace, so it's not the place for testing. The difference is that here we have a fairly recent DSL "box" (modem) while my home cable modem - where oaks fails to update - is probably a 10 years old design.

jldeon, 2016-03-03:
I've been digging into a dozen or so packet captures to try and figure out the `SOCKET READ TIMEOUT` issue.  It seems like the TCP stack on the ESP is not particularly great at handling packet loss.  Every time it fails, I see a pattern of packets dropped and multiple retransmissions of old packets and ACK packets going back and forth.  Eventually there hasn't been valid data in long enough that the Oak gives up on the connection.  

I've been tuning a number of parameters on a custom https server, and that's running here:

```
set
37
{"first-update-domain":"52.37.37.115"}
set
90
{"first-update-fingerprint":"17 08 7A 72 63 78 E9 71 53 50 65 85 16 07 C1 03 9C AA 39 D0"}
```

It's on an Amazon AWS micro instance, and should be able to handle at least a few concurrent connections at once.  

Expect the download to take around 3 minutes or so.

You'll want to do everything you can to ensure a good connection to the internet, as packet loss tends to be fatal.  I suggest dropping your router to B only (on the 2.4GHz band) if you can.  

If you do end up with a `SOCKET TIMEOUT` error on the Oak, you're highly likely to get a few `NO CLIENT` errors until the socket times out on the server side, as per my previous long-winded comment.  

I went from constant `SOCKET TIMEOUT` errors (nearly 100%) to only very occasional errors with this setup.  If this ends up working, I'll share the custom server code :)  Otherwise, it's back to the drawing board for part 3.

jldeon, 2016-03-03:
My Oak's been flashing from my custom server for a while now, I'm up to 37 successful transfers and 3 failures.  Much better than 0 successes last night :)

DarkLotus, 2016-03-04:
Just tried your setup out, First attempt failed at around 70-80% with socket timeout.
Second attempt failed a little earlier around 60% with socket timeout.
Third attempt failed a bit further than the second.
No "NO CLIENT" errors though.

jldeon, 2016-03-04:
@DarkLotus do you have only B enabled on your router?  I've been testing with B/G/N and getting around 30-50% failure rate.  B only is closer to 90% or so.

DarkLotus, 2016-03-04:
Nope, im using B/G/N but from my own tests it made no difference with mine or eriks servers. 

Can't disable G/N right now ( Got long running tasks connected over wifi at the moment ) though will be able to in a few hours time.

jldeon, 2016-03-04:
B-only has been about the change that made the most impact in my testing so far.  Can you keep attempting it?  Would be good for just load testing if nothing else...

I can see your connections to the server and the failed transfers.  I'm busy at the moment, but I may be able to packet capture your attempts at some point, although I bet they're pretty much the same as mine.

DarkLotus, 2016-03-04:
Will do, little busy here as well but will keep pushing some attempts thru, will let you know once I've swapped over to B

DarkLotus, 2016-03-04:
Your servers returning internal error 500 btw. Which gives Update Success and then soft bricks the oak.

jldeon, 2016-03-04:
@DarkLotus Yeah, I just found and fixed a pretty major bug with concurrent connections.  Apologies for any inconvenience.  Your attempts to flash helped me find it :)

DarkLotus, 2016-03-04:
I think what ever you did might have solved it, 2/2 worked just now.

jldeon, 2016-03-04:
Nice!  I saw your attempts go through successfully.  

I've got to sleep now, but I've set up 2 Oaks on my desk in the "silent" debug firmware loop that are flashing continuously.  I've had 1-2 failures over 25 or so attempts.  I'll leave everything overnight and check it in the morning.  

Feel free to abuse the server in the meantime...

jldeon, 2016-03-04:
Overnight testing is complete, the server reports 713 total transfers, with 37 failures (so roughly ~680 successes).  I don't have serial logs to see the results from the client side, since my computer eventually fell asleep.  The Oak failure count is probably at least a bit higher.  

I'll continue refining and testing today, but I could still use more testers pointed at the server...

DeuxVis, 2016-03-04:
@jideon I will test your server tonight. Weekend finally arriving ! Note that my modem/router doesn't allow to disable B G or N.

@DarkLotus Is your server with reduced MTU still available ? I might test it as well.

fri-sch, 2016-03-04:
I previously tried to use beta 0.9.3 and the update worked pretty well.
Now I thought I'll go for 0.9.5, restored my Oak and tried the update a few times without success.
Then I read through the comments above and as it seemed like @jldeon has a server running with some improvements, I thought I will give it a try.
Result: Update worked on the first try

If more testers are needed, I think I could set up an Oak in the endless update loop and let it run for some hours...

jldeon, 2016-03-04:
@fri-sch sure, if you can test and capture your serial output for a while, that'd be a useful data point.  Any info about your router and your connection to the internet would be helpful as well.

I've got 2 oaks running in a continuous loop against my server testing today.  Server logs indicate around 400 successes (including fri-sch's attempts) and around 61 aborts before the file was completely transferred.

I'm still refining things as I have time.  I tested a bit with my Android phone as a wifi hotspot - data coverage in my area is pretty spotty.  I was only successful about 20% of the time.  I've captured some traces, but I'm not sure how fixable the issues are going to be from the server side.

epatel, 2016-03-04:
@jldeon Great work! I think the hotspot test sounds interesting, if one could get a small server running on the hotspot device (well, not sure that is really within the requirements) that could be a way to minimize packet loss. The package only needs to travel over the one "link". Lowering the MTU would increase number of packages, and could maybe increase the probability for packet loss.

fri-sch, 2016-03-04:
@jldeon the update loop is running for some time now and I will leave it on for the night.
My router is an AVM FritzBox 7362 and wifi is setup for b+g+n and uses channel 4.
Let me know if you need any more info...

jldeon, 2016-03-04:
@epatel Yeah, I was considering the possibility of deploying a small HTTP server on people's device or local network or something.  The problem is SSL.  The factory firmware has the domain and thumbprint of the digistump server baked in, and it seems like it can only be changed via serial (which not everyone has).  

I'm not an SSL expert, but I believe that means we'd have to distribute copies of the key plus have people override their DNS to point to their own local copies.  In addition to being painful, it destroys any security SSL brought to the update process.  

@fri-sch Any stats on your success/fail rate?  I'm wondering if it's any better or worse than my 2 Oaks.  Thanks for helping test!

epatel, 2016-03-04:
@jldeon Ah, thought there was some flexibility on setting the loading params as they described. I thought it maybe could be possible to do self signed certs for the hotspot device local address. But then, could it be possible to make a buffering proxy/router? The goal I am thinking of is to eliminate any package loss from wan transport.

digistump, 2016-03-04:
@jldeon, @DarkLotus - awesome work so far, many thanks for all the effort
you are putting into this!

And thanks for all of those helping test as well!

@epatel, @jldeon - The domain, url, and thumbprint can all be changed over
http as well, so those can be changed by the config web app if needed. Of
course, a remote server is preferred but certainly a remote server setup
that was as good as it possibly could be, and some sort of local solution
for those it still didn't work for seems like it could be the best of both
worlds.

digistump, 2016-03-04:
Maybe a self-signed cert for 127.0.01 with an app wrapper that download the
latest firmware and latest config web app and then runs a local server and
presents the web app to the user?

fri-sch, 2016-03-04:
@jldeon: I had a quick look and for the first 108 updates I have 70 successful and 38 failed.
Out of the 38 failures 26 were `COULD NOT CONNECT TO UPDATE SERVER` (so the update didn't start) and 12 were `SOCKET READ TIMEOUT` while updating.

And I noticed, that most of the failures occured while the wifi network was used by other devices, that produced additional traffic.

I'll be back tomorrow with more info...

jldeon, 2016-03-04:
@digistump Oooh, I didn't realize!  My "custom server" is just a bit of python.  It should be possible to deploy it in such a way that people can run it locally on Windows or Linux.  Once we have the kinks worked out, we can work on packaging it.

Do you have a list of all the parameters we can change via the setup app?  There might be something else in there we can use.

jldeon, 2016-03-04:
@fri-sch That sounds about in line with my testing.  The `COULD NOT CONNECT TO UPDATE SERVER` errors probably follow the `SOCKET READ TIMEOUT`s.  The looped updates run fast enough that the server socket doesn't have time to close before the next attempt occurs.

digistump, 2016-03-04:
claim code ("cc"), device-id, meta-id, meta-data (both metas
unused), first-update-domain, first-update-url,
first-update-fingerprint, server-address
(particle server), server-address-type, server-public-key

are the things that can be set - so probably not much else useful there

Any idea what kind of load the little python server can handle?

Ideally it would be if I made the config app do something like this:

Try regular update server (whatever works best apache 2.2, etc with some
tweaked settings per your findings)
If failed, try special update server (your python server still remote)
If failed, direct to download package for local server.

Thoughts?

Thanks!

jldeon, 2016-03-04:
@digistump Hmm, yeah.  I was hoping for a timeout or wireless radio mode parameter.  Rats!  What's the format of the request to change the fingerprint and domain parameters?

The Python server is based on Twisted, and so it ought to scale reasonably well under load.  I can try load testing it a bit to see how many concurrent connections work before we start to throttle, although it's probably going to vary depending on the system running it.  Currently the AWS instance it's on is a micro.

digistump, 2016-03-04:
same json as over serial - just as a raw json post to 192.168.0.1/set

sounds like your server should be fine - we probably need to be able to
handle like 50 concurrent, not 5000 or anything

jldeon, 2016-03-04:
OK.  I'll start working on a standalone, downloadable version for local flashing to compliment the "backup" server.  

When you guys need my code, let me know.  I haven't published it yet so that someone else doesn't totally steal my hard work (:P) but I plan to open source it and/or give you guys whatever license on it you want.

jldeon, 2016-03-05:
I have a working standalone server based on my "improved" https server in python.  It needs a bit of cleanup, but hopefully tomorrow I can put a ZIP file and some instructions up for folks to test out.

fri-sch, 2016-03-05:
@jldeon You're right, the `COULD NOT CONNECT TO UPDATE SERVER` errors follow after `SOCKET READ TIMEOUT`, so I won't count them in the following summary of my overnight test loop:

update runs: 574
update ok: 426
update failed because of `SOCKET READ TIMEOUTS`: 148
success rate: ~74%

[26 more replies]

closed 2016-03-19 by digistump