digistump/OakCore #56: Possible to use Arduino Uno as a 3.3v UART?

jwhendy, 2016-03-16:
Greetings,

I have two oaks that successfully updated and accepted flashing from `arduino` via particle.io, however upon some subsequent activities, they've stopped staying connected to wifi. I'd like to try restoring them, but don't have a 3.3v UART (I have one on order now, but am impatient!).

I wondered about using an Arduino Uno for this, and thought it might be worth posting here since I'm guessing a majority of users will have an Arduino, while many won't have a 3.3v USB UART adapter. I searched around a bit and there's some precedent for this:

- [using voltage divider](http://forum.arduino.cc/index.php?topic=16846.0)
- [using voltage divider](http://stackoverflow.com/questions/27037377/arduino-uno-to-connect-with-serial-3-3v-device)
- [says it just works as-is (5V)](http://forum.arduino.cc/index.php?topic=283043.0)
- [uses 3.3v regulator through a 5V UART](http://electronut.in/an-iot-project-with-esp8266/)
- [uses a 3.3v LDO, but also quotes the datasheet that the ESP8266 is 5V tolerant](https://www.element14.com/community/groups/internet-of-things/blog/2014/11/19/esp8266-wi-fi-arduino-upload-to-xively)

> All digital IO pins are protected from over-voltage with a snap-back circuit connected between the pad and ground. The snap back voltage is typically about 6V, and the holding voltage is 5.8V. This provides protection from over-voltages and ESD. The output devices are also protected from reversed voltages with diodes. (p17-18 of [the datasheet](https://www.adafruit.com/images/product-files/2471/0A-ESP8266__Datasheet__EN_v4.3.pdf))

I don't know if "all digital pins" encompasses everything required to flash the Oak via UART, nor do I know if this statement applies to all of the Oak's circuitry...

Anyway, I wanted to find out thoughts on if this is possible as it might make programming/debugging a lot more accessible and prevent the frustration of ordering a part just for use with the Oak (I'll have little use for it other than this purpose).

Thoughts?

P.S. If someone could confirm it's feasible, I'd be willing to try it and write up a tutorial with pictures. I'd start with the voltage divider method, confirming the divider with a multi-meter before hooking things up to the Oak. I'd *possibly* consider risking an Oak with the straight TX/RX pins off of the Uno if someone could let me know if the Oak is 5V tolerant.

DeuxVis, 2016-03-16:
Someone else on the forums reported using that successfully, and I don't see any divider resistor or level shifter in his setup picture : http://digistump.com/board/index.php/topic,2036.msg9266.html#msg9266

jwhendy, 2016-03-16:
Thanks for that @DeuxVis ! I can confirm it works, though the arduino's labeling is a bit misleading (I initially thought it was backwards, but I can't fathom that...). Namely the pins are labeled:

- `TX -> 1`
- `RX <- 0`

So, per [the instructions](https://github.com/digistump/OakRestore) I:

> Connect Pin3/RX to TX and Pin4/TX to RX, GND to GND.

I'm thinking the labels meant that pin 1 on the arduino is to *receive* the TX of another device, and vice versa. I think the labels look like pin 1 is the arduino TX, not the pin *for* the TX. In re-looking at the picture you linked to, it looks the same (p1 on arduino is definitely going to p4/TX on the oak).

In any case, really happy this works!

---

Is this enough of interest for a small tutorial for others who want to flash with an Arduino? I could take pictures of my setup while I have it up (with and/or without a voltage divider).

closed 2016-03-16 by jwhendy

jwhendy, 2016-03-17:
Just to confirm, the pin labeled `RX <- 0` on the arduino *definitely* goes to pin 3/RX on the oak. I'm not sure who's backwards, but it's definitely confusing. This setup worked for me:

```
| arduino | oak         |
|---------+-------------|
| gnd     | gnd         |
| 3.3v    | Vin[1]      |
| TX -> 1 | pin 4/TX    |
| RX <- 0 | pin 3/RX    |
|         | gnd to pin2 |

[1] Note: micro USB for power is fine as well; just make sure
in either case the grounds of the oak and arduino are connected
```

I have pictures of the setup. I assumed the oak pins were correct and thus used the voltage divider on the arduino p0/RX with TX going straight to TX on the oak. Like so:

```
arduino
(RX <- 0) ---- 220 ohm ---- oak p3/RX
                 |
                 |
	      1k ohm
                 |
		 |
arduino gnd ----------- oak gnd
```

I originally used a 220 and 470 ohm for the divider (should have been roughly 2/3 and was measuring 3.27V with the 5V on the Arduino. My attempts to run the `esptool.py` were failing to connect via the divider, but would with straight connections. I replaced the 470 with the 1k ohm, and that worked. Not sure what happened there?

jwhendy, 2016-03-17:
Bah! Finally dawned on me! So for an actual UART, the instructions are correct.

```
Computer             Oak
--------             -----
TX ----------------> RX
RX <---------------- TX
```

But I'm connecting to the oak *through* an arduino board, which means the first leg of the trip is like this:

```
Computer             Arduino
--------             -------
TX ----------------> RX
RX <---------------- TX
```

To use the computer's TX signal as an example, we would want this to reach the RX pin on the Oak. Since that signal comes into *RX* on the arduino, we want the RX port on the oak connected *there*. The computer is just passing it's signal on through:

```
Computer    Arduino      Oak
--------    -------      ---
TX ---------> RX ------> RX
RX <--------- TX <------ TX 
```

My fatal flaw was in envisioning the *arduino* talking to the oak! The arduino would normally "talk" out of TX, but it's not doing any speaking... *We* want to talk through that pin from the oak to the *computer*, hence hooking up the oak TX there (and vice versa with RX).

Whew. That would have kept me up if I didn't explain it through typing it. Hope someone finds this at all helpful and it's not just me who got this wrong!

As it turns out, I was correct in applying the voltage divider to the signal going into RX, as that would have been the computer's [presumably] 5V signal.

DeuxVis, 2016-03-17:
Yeah I'm pretty sure a nice visual tutorial on this use case would help : There is probably a fair amount of people out here who already have an arduino but no USB/serial adapter, and reading your last posts here, I believe some clear instructions on the wiring would be welcome.
Thanks for your time if you write one.

jwhendy, 2016-03-17:
@DeuxVis [Started one](http://digistump.com/wiki/oak/tutorials/serial_through_arduino)! Got a little carried away with some background on voltage dividers and now I need to get to work, but I'll update with a picture of my setup later today, possibly with a Fritzing as well.

While it's probably not necessary, I thought the voltage divider method should be the "official" recommendation, with citing examples of where it was done with direct connections as a has-worked-but-use-at-own-risk footnote.

Are the explanations above reasonable? If you think they make sense, I'll leverage the gist of how it "clicked" for me. If you think my previous post was still confusing, I'll try to think of another way to lay it out/explain why TX/TX and RX/RX between the oak and arduino.

Thanks for the encouragement! I actually quite enjoy writing tutorials :)

DeuxVis, 2016-03-17:
Yes the above explanations are fine. Maybe introducing by explaining the arduino board is only used as a passive bunch of wires, and doesn't need an active behaviour (no programming) would be good.

jwhendy, 2016-03-17:
@DeuxVis Will do. In fact, I used mine with no board (just how I'm used to when I'm flashing an atmega328 on a breadboard through the Uno). One could jump resent to gnd as well. In addition to the explanation, hopefully that will get the point across (no brains in it).

jwhendy, 2016-03-18:
Tutorial [finished](http://digistump.com/wiki/oak/tutorials/serial_through_arduino).

DeuxVis, 2016-03-18:
Excellent ! And you've gone beyond expectations with that pot reading sample sketch. Thanks.

emardee, 2016-03-18:
jwhendy,  Looking good from here. Only suggestion I'd have would be to drop the link to your tutorial down 5 places on the Oak wiki, so it sits between the link to the tutorial for bus pirate, and the one for loading firmware by serial. Seems that would be the best place for the link.

jwhendy, 2016-03-18:
@emardee Done! Actually missed those, so hopefully mine isn't too redundant...

rkinnett, 2017-01-09:
@jwhendy I wanted to give you a heads up that I massively overhauled the [Serial Comm through Arduino](http://digistump.com/wiki/oak/tutorials/serial_through_arduino) page which you originated.  I would be glad to hear what you think of it.  Also left an inline fixme note about needing to fix the Fritzing diagram; not sure if I'll get to that or leave it for the next passerby.

I wasn't aware until after I rewrote the page that @digistump had asked for a specific format.  I think in this case the technical nature of the material warranted a non-standard structure, so I'm in favor of keeping the new structure but am open to suggestions if it needs to conform.

Cheers-

jwhendy, 2017-01-09:
@rkinnett Looks good to me. Looks good to me! It actually seemed like a lot of re-arranging/supplementing vs. an overhaul, per se, but I admit to primarily skimming. I looked and don't seem to have that Fritzing diagram as a starting point but could probably generate one if we want to illustrate the voltage divider.

jwhendy, 2017-01-09:
@rkinnett  Went to tackle the fritzing and actually do have more comments now.

- why recommend the bi-dir converter vs. just a regular 3.3V capable [USB UART](https://www.amazon.com/HiLetgo-Ft232rl-Serial-Adapter-Arduino/dp/B00IJXZQ7C)? My original intent was to give folks a way to communicate via serial *without* such a device since most will already have an Arduino but probably not a UART. Feel free to defend; I just was surprised that this would be the most recommended way given the original idea behind the tutorial (how do I do this serial stuff with what I have around). Once one starts buying logic converters, might as well ditch the Arduino (at least in my mind).

- how did you arrive at the 1k suggestion for the current limiting method? It's probably fine; I'm just curious on what you might be targeting with respect to an Oak input current. I thought the overall output of the Arduino was 500mA (so believe Tx will be <= this). Is there documentation about whether or not the ESP8266 can handle 500mA? I looked in the data sheet and only see 12mA max *output*, but nothing about dangerous input to watch out for.

- I think the move to a divider is likely a good thing. In re-checking, I found it interesting that the [current version](https://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf) of the ESP8266 documentation *does not* contain the former quote about over-voltage protection (found in [v4.3, 2015](https://cdn-shop.adafruit.com/product-files/2471/0A-ESP8266__Datasheet__EN_v4.3.pdf)):

> All digital IO pins are protected from over-voltage with a snap-back circuit connected between the
pad and ground. The snap back voltage is typically about 6V, and the holding voltage is 5.8V.

Attaching Fritzing; let me know what you think. I can upload the .fzz file if you'd like to re-do or whatever. Hope that helps!
[image]

rkinnett, 2017-01-13:
@jwhendy thanks for reviewing and being open to changes.  All good points.  

Nice job on the Fritzing diagram; you nailed it.  I went ahead and posted it in the article.

Regarding the option of using a level shifter, I didn't intend for that option to be the primary recommendation, but you're right, that's the way it came across.  Level shifters are becoming more common these days, so I figured it's worth noting as an option for anyone who happens to have one (but didn't have a 3V USB/serial converter).  I updated the subsection titles to call these Option 1 and Option 2, and also added notes about this tutorial focusing on the voltage divider option because it uses common components.  Let me know if that's more clear, or feel free to edit.

Good question re the 1K value for the current limiter option.  1K is fairly common for connecting TTL things.  See RC Navy's post [here](http://digistump.com/board/index.php/topic,1596.0.html), for example.  The requirement isn't so much on input current, but rather on the output current of each device, and it's good practice to consider what happens if the devices startup in opposite states.  Following this logic, we may want to recommend a resistor between Oak/Arduino Tx as well.  The Arduino Tx pin can source 40mA max, which argues for at least 125 ohm between Ard "Rx" and Oak Rx.  Then again, just because the Arduino pin _can_ source 40 mA doesn't mean it _should_ or needs to; might as well limit it further for margin and efficiency.  The current limiter resistor and the voltage dividers can go much higher than 1K to reduce current bleed without losing signal since the input impedances of the UARTs are presumably in the MOhm range.  That's basically why I increased the resistors in the parts list as well.  The values that you had originally were safe, and we don't really need to optimize current bleed (it's not like we're powering both the Arduino and Oak from a li-po to reflash the Oak!), so we can revert if you prefer.

The data sheet that you pointed to in your last post (v4.3) does still describe overvoltage protection on the GPIO pins, at the bottom of pg 17, and the Tx/Rx pins double as GPIO, so they should include the protection as well.

Just found an [interesting note](http://www.ba0sh1.com/2016/08/03/is-esp8266-io-really-5v-tolerant/) about the CEO of Espressif officially announcing that the ESP8266 is 5V tolerant.  Hmm, does that change our recommendation?

rkinnett, 2017-01-13:
@jwhendy sorry, I misread your note about the version of the data sheet.  The "[current version](https://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf)" you posted above is for the ESP8266**EX**.  I don't know if the Oak's ESP-12-F board uses the same chip (?).

jwhendy, 2017-01-15:
@rkinnett Thanks for the correction on the board! I completely missed the "EX" at the end. I can't seem to find a data sheet that lists the "regular" ESP8266 *and* the over-voltage protection, though. Check the v4.3 version... I just now caught that it says "EX" on the title page too. Then again, I'm on the espressif site and using their [document map](https://espressif.com/en/support/explore/get-started/document-map/esp8266) to hunt. When I click the ESP8266 link under Documentation -> Hardware -> Data Sheet, I get [this](http://www.espressif.com/sites/default/files/0a-esp8266ex_datasheet_en_1.pdf) which also says it's for the EX. I can't get any hits for "protect", "5V", "tolerant" or similar from the 2015 version about GPIO pins, pg17.

Thanks for the explanation on the 1k resistor. I'm happy with that suggestion and was more just curious where it might have come from.

All sounds great and thanks for your effort on the documentation! Always happy to have more eyes and hands looking at things. I think you did a great job!

susumakeit, 2018-07-04:
There is another way to us "TTL level converters 5V -> 3.3V"