digistump/OakCore #65: Tone implementation

jackygurui, 2016-05-12:
Hi @digistump just wondering if there will be an implementation of https://github.com/digistump/OakCore/blob/master/cores/oak/Tone.cpp any time soon?

digistump, 2016-05-13:
Yes - likely soon - the main ESP8266 project has recently added an implementation, which you can use (https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Tone.cpp) but there are some reports of major bugs with it (that you can't turn it off, can't change the tone) - so we'll wait for it to be a bit more stable before we merge it - leaving this open though so I don't forget

micahwedemeyer, 2016-07-20:
👍  for tone support. I'm running a buzzer and playing some basic sound on it via PWM, but it would be great to be able to use tone.

digistump, 2016-07-25:
The next release will have this integrated - in the mean time you can replace your Tone.cpp for the Oak with the latest one and it "should" work: https://github.com/digistump/OakCore/blob/master/cores/oak/Tone.cpp

ekettenburg, 2016-10-11:
in 1.0.6

closed 2016-10-11 by ekettenburg